s60sc/ESP32-CAM_MJPEG2SD #1: Modifications to use Active-High PIR

jameszah, 2020-02-21:
Excellent program!

Here are the mods for an active high pir, for anyone using one of those.
Three small changes in mjpeg2sd.cpp file.

> //jz pinMode(PIRpin, INPUT_PULLUP); // pulled high for inactive
>   pinMode(PIRpin, INPUT_PULLDOWN);    // mod for active high pir
> showInfo("To record new MJPEG, attach PIR to pin %u or simulate by powering pin %u", PIRpin, PIRpin);

>   //jz mjpegStatus = (digitalRead(PIRpin)) ? 0 : 2; // in case PIR off missed
>   mjpegStatus = (!digitalRead(PIRpin)) ? 0 : 2;     // mod for active high pir

>   //jz mjpegStatus = !digitalRead(PIRpin);
>   mjpegStatus = digitalRead(PIRpin);        // mod for active high pir

s60sc, 2020-02-21:
Thanks for your comment and the reference from your site.
I would be interested in knowing what SD card you used and what frame rates you achieved.

jameszah, 2020-02-24:
Hi, haven't done anything rigorous, but I was getting an error that might be interesting.
Using your program, switched to high-active, and pinned to 3.3v
Both chips quick formatted 32k block

First 
 LEXAR 32 GB 300x circle10 A1 U1 V10 HC1
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Record VGA mjpeg, width: 640, height: 480, @ 30 fps
01:44:26.275 -> ..................................................
01:44:29.636 -> ..................................................
01:44:32.642 -> ..................................................
01:44:35.592 -> ..................................................
01:44:38.509 -> ..................................................
01:44:41.308 -> ..................................................
01:44:44.086 -> .......................
01:44:45.425 -> ******** MJPEG recording stats ********
01:44:45.425 -> Recorded /20200224/20200224_014424_VGA_17_19.mjpeg
01:44:45.460 -> MJPEG duration: 19.2 secs
01:44:45.460 -> Number of frames: 324
01:44:45.460 -> Required FPS: 30
01:44:45.460 -> Actual FPS: 16.9
01:44:45.460 -> File size: 6.01 MB
01:44:45.460 -> Average frame length: 19444 bytes
01:44:45.460 -> Average frame process time: 4 ms
01:44:45.460 -> Average frame storage time: 47 ms
01:44:45.460 -> Average SD write speed: 400 kB/s
01:44:45.460 -> File open / completion times: 7 ms / 36 ms
01:44:45.460 -> Busy: 88%
01:44:45.460 -> Free heap 673724 bytes, largest free block 597820 bytes
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Second
 Dollar store chip called EAGET 16 GB HC1 circle10 U1 TF CARD
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 Record VGA mjpeg, width: 640, height: 480, @ 30 fps
01:48:29.901 -> ............................................E (45511) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
01:48:34.475 -> E (45511) diskio_sdmmc: sdmmc_write_blocks failed (263)
01:48:34.475 -> ......
01:48:34.653 -> ..................................................
01:48:36.618 -> ..................................................
01:48:38.625 -> ..................................................
01:48:40.634 -> ..................................................
01:48:42.622 -> ..................................................
01:48:44.635 -> .................................E (61958) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
01:48:50.899 -> E (61958) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
01:48:50.934 -> E (61959) diskio_sdmmc: sdmmc_write_blocks failed (263)
01:48:55.923 -> E (66965) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
01:48:55.923 -> E (66965) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
01:48:55.923 -> E (66966) diskio_sdmmc: sdmmc_write_blocks failed (263)
01:49:00.909 -> E (71972) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
01:49:00.909 -> E (71972) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
01:49:00.944 -> E (71973) diskio_sdmmc: sdmmc_write_blocks failed (263)
01:49:00.944 -> 
01:49:00.944 -> ******** MJPEG recording stats ********
01:49:00.944 -> Recorded /20200224/20200224_014827_VGA_21_16.mjpeg
01:49:00.944 -> MJPEG duration: 16.0 secs
01:49:00.944 -> Number of frames: 334
01:49:00.944 -> Required FPS: 30
01:49:00.944 -> Actual FPS: 20.8
01:49:00.944 -> File size: 4.55 MB
01:49:00.944 -> Average frame length: 14290 bytes
01:49:00.944 -> Average frame process time: 34 ms
01:49:00.979 -> Average frame storage time: 11 ms
01:49:00.979 -> Average SD write speed: 1179 kB/s
01:49:00.979 -> File open / completion times: 170 ms / 15020 ms
01:49:00.979 -> Busy: 100%
01:49:00.979 -> Free heap 673436 bytes, largest free block 597820 bytes
01:49:00.979 -> ***************************

That second chip is cheap but seems to work fine but the results are odd.  
Those errors - timeouts I think -- but the fps reported higher, which must be wrong.

The frame length for the second is smaller - so I might have been pointing at my desk rather than the monitor.  So that confuses matters.

Puzzled here about high active PIR causing havoc on pin 12 for rebooting ESP!

s60sc, 2020-02-24:
I added a note to the ReadMe. Pin 12 is used during bootup which will fail if the PIR is high, but this bootup property is not needed and can be disabled by an efuse.
The timings are not completely accurate particularly for short videos, because multi tasking means that 2 activities can share the same time period and so be under reported. The errors will have messed up the stats.
FPS depends on image size which varies according to image complexity.

jameszah, 2020-02-26:
I think my problem was that I had pin 12 tied to 3.3v, which provided lots of current to annoy the chip.  But when I have it tied to a PIR producing 3.3v but tiny current, then everything was fine.

That efuse business sounded a little scary - I have to read more about that.

jameszah, 2020-03-12:
Howdy,

Hardware question for you:

Have you considered trying to run your program without using any psram?

I had the idea that I could improve the speed, and overcome random sd delays by having a queue of frames, than would be written to the sd when if was free.  I could reserve frame buffers for about 15-20 vga frames within the camera system.   But whenever the queue filled up, it would never empty, even though the sd went back to its usual writing speed, which was much lower than the frame rate.  

Then I discovered that a vga photo (esp_camera_fb_get()) would normally take much less than 1 ms, but after I had 10+ frames inside the camera system ("the psram"), the picture would take 50+ ms.  So the time was used writing the picture into psram, while independently my avi writer task was waiting to grab a different picture out of psram.  

I think your program just has 1 fb_get outstanding, so the camera drivers are not filling up psram, but I think you have a 512KB buffer where you stack up the frames before writing them.  I think you must be exceeding the sram, but just by a little.  I was trying to use the entire psram, which didn't work at all.

I not sure if the psram is all that much faster than the sd card interface, although perhaps a little more predictable without the occasional randomness of the sd???

Any thoughts?

s60sc, 2020-03-13:
Hi
The number of frame buffers needed increases with reducing size, so smaller images need more buffers to reach the cluster size before they are written to sd. The 512k buffer size is only there for highest quality uxga images and because there is plenty of psram. If you just want to use internal ram with small images then a 64k buffer (for double buffering 32k cluster) would do. The bottleneck is the slowest component - the sd card, it doesn't really matter that psram is slower than internal ram when you use tasking to time share as memory activities get scheduled during the waits for SD.

closed 2020-03-13 by jameszah