s60sc/ESP32-CAM_MJPEG2SD #2: Creating MJPEG from existing JPG files

tobozo, 2020-02-26:
Just discovered your project through the github suggestion panel and it looks great !

Looking at the code I was wondering how to use it as a basis to mjpeg-encode compressed images that aren't originated from an OVxxxx camera e.g. from a tft.readRect + bodmer's JpegEncoder or from normalized jpeg files found on the SD Card.

I'm also thinking about chaining this to the [imgur uploader](https://github.com/tobozo/ImgurUploader) since the mjpeg format is supported by the imgur platform.

The reason why I'm investigating this is to produce mjpeg version of tft screen capture on ESP32 such as the one you can see on [this project page](https://github.com/tobozo/ESP32-Tesseract), so it can spare the hassle of building the GIF manually.

Please let me know whether you'd like to inherit of those changes (and I'll make a fork + PR) or if it's okay with me creating a new repo (obviously with credits to your project in the readme).

s60sc, 2020-02-26:
Thanks

In ```processFrame()```, the key line is:
  ```memcpy(SDbuffer+highPoint, fb->buf, jpegSize);```
which copies out the jpeg from the camera frame buffer ```fb->buf```
so you could load SDbuffer with your own jpegs using a separate synchronised task.

For existing jpegs use a tool such as ffmpeg. Remember that mjpegs do not have an inherent frame rate so you may want to wrap them in a container such as avi.

Probably simplest for you to make a separate repo

tobozo, 2020-02-26:
thanks ! I'll keep you updated

closed 2020-09-17 by s60sc