s60sc/ESP32-CAM_MJPEG2SD #4: Can't connect Pin 12  PIR until after boot-up

LuapGG, 2020-06-19:
Thanks for this great project! Use it to capture birds (using a PIR)  in the garden and works a treat!
I was just wondering whether there is any code that could be added so that PIR can remain connected on Pin 12  at boot-up  ?

s60sc, 2020-06-19:
Thanks.

Pin 12  is read on bootup before any code is executed, so no.

You can permanently disable the unnecessary bootstrapping behaviour of pin 12. It is not as scary as it looks.
https://github.com/espressif/esptool/wiki/espefuse#setting-flash-voltage-vdd_sdio

Alternatively if you dont need the led lamp, change PIRpin to 4. 
Or if you are OK with a bit of soldering, use pin 33 attached to the small red led.

LuapGG, 2020-06-20:
I went for soldering pin 33 and works great.
Thanks again!

closed 2020-09-17 by s60sc