s60sc/ESP32-CAM_MJPEG2SD #6: audio support?

robertstass, 2020-06-28:
Do you think it'd be feasible to add support for a microphone? Either externally connected or using one of the esp-cam like boards that contain onboard i2c microphones (TTGO ect)? I'm thinking about attempting this but I think it might be a bit above my technical ability. Would there be sufficient pins?
Thanks

s60sc, 2020-06-28:
If it is just as a sound detector you could use a ky-038 connected to pin 12 or 33. 
If it is to record synced audio then no chance.

s60sc, 2020-07-05:
Thinking about this further, I'm going to try implementing audio record & playback as an intellectual exercise, but it could be a while or not at all that anything is released. If you want to get involved then my idea is to record 8 bit PCM in chunks to SD then playback to browser on the fly as 16 bit PCM in a WAV. A further enhancement would be to combine PCM and MJPEG into an AVI on the fly when uploading by FTP. Only one ADC pin (pin 33) is needed to record 8 bit audio, but the quality may be too bad. I won't know until I try a MAX9814 module I've ordered from China.

robertstass, 2020-07-05:
Glad to hear you're thinking about this. Please keep me updated on how you get on though no worries if it can't be done.

s60sc, 2020-09-27:
Limited microphone support available - look in avi.cpp. ESP32 analog input is noisy.

closed 2020-09-27 by s60sc

ram-123455, 2023-08-21:
I am not able to record the audio with and without video using Seeed Xiao ESP32S3 Sense?