s60sc/ESP32-CAM_MJPEG2SD #9: GPIO33 is available?

Ab-Tx, 2020-07-12:
Hello, first of thanks for this amazing project you pulled off!
From the schematic of the ESP32-Cam the GPIO33 only connects to an LED (the small one on the back of the board), so it is safe to use it for something else, right?
If it can be used it will be handy to control a fan based on the ESP32 temperature.

s60sc, 2020-07-12:
Thanks
Yes you can repurpose pin 33. I've tried it for other things such as an LDR, and a microwave sensor, I am currently investigating using it for a microphone. 
I use pin 3 for a DS18B20 temperature sensor - it doesn't interfere with pin 3's use for the serial interface.

Ab-Tx, 2020-07-12:
Nice, using pin 3 is a really clever idea since the ESP32 won't be receiving any info from the serial interface.

closed 2020-07-12 by Ab-Tx

s60sc, 2020-07-12:
Bear in mind pin 3 may only be usable for input. Pin 1 may be usable for output - I haven't tried.

Ab-Tx, 2020-07-12:
Indeed, I'm guessing the serial monitor may not show useful information if pin 1 is used as an output.
By the way, I just noticed a global float type variable named "ambientTemp, is the code already reading the internal temperature from the ESP32?

s60sc, 2020-07-12:
The internal temp is no longer available on later version ESP32s.
I use a DS18B20 on my own camera project, but I've not included the code in this repository, but I could add it when I do another update.