s60sc/ESP32-CAM_MJPEG2SD #10: Can not connect to SSID with spaces

burkadius, 2020-08-01:
Hello s60sc, first of all thank you very much for publishing your program code, it is exactly what I was looking for, I especially like the possibility to delete files on the SD card. Unfortunately I am not a programmer and have only very limited knowledge.  But anyway, the installation of the code is successful and I can connect to the camera in AP mode. But if I enter my SSID either in the code (myConfig) or under 192.168.4.1 there is no connection to the router. I get in the monitor of the IDE the info "connectet to 0.0.0.0 . Then I found out that it works without spaces in my SSID.
Can you give me a hint what I am doing wrong.
Best regards
Burkhard

s60sc, 2020-08-01:
Thanks
How long is your ssid, if longer than 20 chars then increase ST_SSID[20]

burkadius, 2020-08-01:
Yes, You are right, my SSID is 22 chars long and also my pass (21 char). But there are no better results after changing to 23 char for both

s60sc, 2020-08-01:
@gemi254 who contributed this code may be able to help

burkadius, 2020-08-02:
ok, thanks for this link, I will ask gemi.

gemi254, 2020-08-05:
> Hello s60sc, first of all thank you very much for publishing your program code, it is exactly what I was looking for, I especially like the possibility to delete files on the SD card. Unfortunately [...]

Spaces not handled correctly without url encode url. Also ssid field size was too small.
Laters version 
https://github.com/gemi254/ESP32-CAM_MJPEG2SD
has fixed the problem using url encoded requests so it will work correctlly…
If you wish compile with 
Tools->Core Debug level -> Verbose 
and check serial monitor..

drmartin1234, 2020-09-15:
I agree with Burkhard, excellent software.  However, I think I have a follow on issue.  My SSID has a period i.e. a "." at the end.  The cam will not log on to the router.  I have many cams that log on to the Espressif supplied web server with no problems.  I also have many other ESPs doing MQTT with no problem logging on.  I get the same, "I get in the monitor of the IDE the info "connectet to 0.0.0.0 ." after it fails.  What do you think?  I can get rid of the "." its just going to be a pain reflashing all those ESPs.  But your software is worth it.  Thank you!

s60sc, 2020-09-15:
I implemented a fix that should address this but didnt upload it as I was working on further changes that I haven't got round to finishing.  I've  uploaded the fix now.

gemi254, 2020-09-17:
Some special chars lkie dot not handled correctly in encoded url function.
Laterst version
https://github.com/gemi254/ESP32-CAM_MJPEG2SD
Fix this

s60sc, 2020-09-17:
Uploaded a fix for dots. Will subsequently make function generic when do merge

closed 2020-09-17 by s60sc