s60sc/ESP32-CAM_MJPEG2SD #13: error during compiling

burkadius, 2020-09-25:
Hi, I'm trying to install the latest release and get this error information 
Arduino: 1.8.12 (Mac OS X), Board: "AI Thinker ESP32-CAM"

sketch/app_httpd.cpp.o:(.literal._ZL14stream_handlerP9httpd_req+0x48): undefined reference to `openSDfile()'
sketch/app_httpd.cpp.o:(.literal._ZL14status_handlerP9httpd_req+0x118): undefined reference to `readDStemp(bool)'
sketch/app_httpd.cpp.o: In function `stream_handler(httpd_req*)':
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: /Users/marlisdrechsler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi
/Users/marlisdrechsler/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/bits/shared_ptr_base.h:127: undefined reference to `openSDfile()'
sketch/app_httpd.cpp.o: In function `status_handler(httpd_req*)':
/Users/marlisdrechsler/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/bits/shared_ptr_base.h:127: undefined reference to `readDStemp(bool)'
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board AI Thinker ESP32-CAM.

The older file is compiling without problems
Best regards 
Burkhard

s60sc, 2020-09-25:
Oops - try latest code base

Burkadius1952, 2020-09-25:
Hi s60sc, compilation is working now, thanks.
But I can not connect to my WLAN. Only the AP connection works. If I'am trying to install the IP, SSID and pass in the AP mode  and save it all, then reboot nothing is happen. the IDE monitor : Camera ready, version 1.7. Use "http:// 0.0.0.0." to connect. Night time 
If I install theWLAN settings in the myconfig.h , line15-17, the compilation stoppt on this point.
Any hints ?
Thanks
Best regards 
Burkhard

s60sc, 2020-09-25:
I think its because you took a build from @gemi254, and I have not yet merged the pull request, so use the @gemi254 build until I have done the merge

s60sc, 2020-09-27:
New merged build uploaded

closed 2020-09-27 by s60sc

Burkadius1952, 2020-09-29:
Hello s60sc. thank you very much for the revised sketch. Now everything works very well. I also tested the temperature sensor and OTA. Very good. I have another question about the SD-Card: What happens when the SD-Card is full, i.e. has no free space left ? Will old data be overwritten then, as for example with a ring buffer or will the recording simply stop ?
Many greetings
Burkhard

s60sc, 2020-09-29:
Good to hear it's working ok.The SD write will fail with an error. You would need to do manual deletion to free up space.