s60sc/ESP32-CAM_MJPEG2SD #14: Offline operation

Ab-Tx, 2020-09-27:
Hello,
Thanks for sharing this new firmware, it is really amazing.
Even the colors of the camera look better now, I have no clue why.

Could it be added an option on the code (or on the webserver) that would make the ESP32-Cam operate offline ?

Also, the code does not clean the SD card automatically, does it? Would you mind if I add that on a pull request? I'm just unsure where this verification should be triggered.

Since this isn't really an issue I'll close it as soon as anyone answers.

s60sc, 2020-09-27:
Thanks
Not sure what you mean by offline - switch off WiFi?
Files can be manually deleted, but not entire card
If you want to write your own code for above for merging then you can make a pull request

Ab-Tx, 2020-09-27:
Yes, switching off WiFi but still being able to record from motion detection and/or PIR sensor.

closed 2020-09-27 by Ab-Tx