s60sc/ESP32-CAM_MJPEG2SD #17: video record without SD

francescot92, 2020-11-16:
hi, your project is very fantastic. 
i'd like to start from your project for record a video without sd and sent it on a cloud. can you suggest me?

s60sc, 2020-11-16:
Thanks

I've not done anything with cloud, but you could combine with cloud code from other esp32 projects. 
gemi254@hotmail.com who also contributes to this project may be able to help.

gemi254, 2020-11-17:
This can be done i think with http post requests in a remote host running http server.. But it will need remote side software.. Just saving the stream should be possible
Do you have a esp32 cloud project in mind.

francescot92, 2020-11-17:
At first, I try to do the same thing that you have done without SD card. I
record the video from the streaming in a buffer in PSRAM and play this
video like a playback.  After that, I would like to download or play this
video in a cloud but I haven't idea how to do it. if you can suggest me how
to do it.
thank you for the reply
regards

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

s60sc, 2020-11-17:
You can't store a video in psram - no room, you either have to stream directly to cloud (may be impractical) or buffer it on SD card first. However as before, I can't help with cloud.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>

francescot92, 2020-11-17:
in your project you have used PSRAM to buffer the camera frames and you
have costructed the MJPEG file really? i'd like to send this buffer in
http. Is it possible?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

s60sc, 2020-11-17:
The browser streaming interface uses http, but this just receives the latest available frame from esp32, if connection is too slow frames are missed by the stream, the complete video is only available on SD card. Not sure I can help any further.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>

francescot92, 2020-11-17:
thank you, my goal is a video composed by few frames for example three or 4
frames because my final project will be:
when the motion will be detected with a PIR the esp32 will make a video
composed by 3/4 frames and will send it on a cloud or anything that allows
me to download or play it. it is possible without SD ?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

s60sc, 2020-11-17:
For 4 frames or less you don't need SD card.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

gemi254, 2020-11-18:
I don't know if you can't store even a tiny video in psram, only some frames.
But project uses PSRAM to buffer the camera frames and stream them over http (or store them in a sd file). To send this buffer in http is the same as stream.
So it is better to store theese frames remotelly, maybe need a server site script (cloud of http) that will remotelly record this mjpeg stream. The recording can be enabled remotelly by esp32 when a motion is detected. It will receive the latest available frames from esp32, but if connection is too slow some frames will be missed by the stream.

closed 2020-12-23 by s60sc