s60sc/ESP32-CAM_MJPEG2SD #18: Multiple definition errors in WiFi sections

lljaxn, 2020-12-28:
I am getting the following errors when trying to compile with fresh install of Arduino 1.8.13 and 

sketch\utils.cpp.o: In function `ipToString(IPAddress)':
C:\Users\lajac\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS\src/FS.h:47: multiple definition of `ipToString(IPAddress)'
sketch\ESP32-CAM_MJPEG2SD.ino.cpp.o:sketch/myConfig.h:212: first defined here
sketch\utils.cpp.o: In function `setWifiAP()':
utils.cpp:(.text._Z9setWifiAPv+0x0): multiple definition of `setWifiAP()'
sketch\ESP32-CAM_MJPEG2SD.ino.cpp.o:ESP32-CAM_MJPEG2SD.ino.cpp:(.text._Z9setWifiAPv+0x0): first defined here
sketch\utils.cpp.o: In function `setupHost()':
utils.cpp:(.text._Z9setupHostv+0x0): multiple definition of `setupHost()'
sketch\ESP32-CAM_MJPEG2SD.ino.cpp.o:ESP32-CAM_MJPEG2SD.ino.cpp:(.text._Z9setupHostv+0x0): first defined here
sketch\utils.cpp.o:(.bss.tmReboot+0x0): multiple definition of `tmReboot'
sketch\ESP32-CAM_MJPEG2SD.ino.cpp.o:(.bss.tmReboot+0x0): first defined here
sketch\utils.cpp.o:(.bss.tmConn+0x0): multiple definition of `tmConn'
sketch\ESP32-CAM_MJPEG2SD.ino.cpp.o:(.bss.tmConn+0x0): first defined here
sketch\utils.cpp.o: In function `checkConnection()':
utils.cpp:(.text._Z15checkConnectionv+0x0): multiple definition of `checkConnection()'
sketch\ESP32-CAM_MJPEG2SD.ino.cpp.o:ESP32-CAM_MJPEG2SD.ino.cpp:(.text._Z15checkConnectionv+0x0): first defined here
sketch\utils.cpp.o:(.bss.dnsAPServer+0x0): multiple definition of `dnsAPServer'
sketch\ESP32-CAM_MJPEG2SD.ino.cpp.o:(.bss.dnsAPServer+0x0): first defined here
sketch\utils.cpp.o: In function `startWifi()':
utils.cpp:(.text._Z9startWifiv+0x0): multiple definition of `startWifi()'
sketch\ESP32-CAM_MJPEG2SD.ino.cpp.o:ESP32-CAM_MJPEG2SD.ino.cpp:(.text._Z9startWifiv+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1

s60sc, 2020-12-29:
cant help with arduino issues

closed 2020-12-29 by s60sc

lljaxn, 2020-12-29:
I was able to get the project loaded by commenting out the duplicate function declarations in myConfig.h

lljaxn, 2020-12-30:
This was caused by me not cleaning the folder \ESP32-CAM_MJPEG2SD which I had used for a previous version from another author