s60sc/ESP32-CAM_MJPEG2SD #19: ERROR: pSRAM must be enabled

lljaxn, 2020-12-29:
Does anyone know how to get around this error? I have tried two different boards "AI Thinker" Clones and both get this error. the pSRAM is on the devices

Camera init OK
SD ready in 1-line mode 
SD card type SDHC, Size: 3796MB, Used space: 0MB, Total space: 3763MB
Loading config..
..
ERROR: pSRAM must be enabled
Unable to continue, SD card fail,

s60sc, 2020-12-29:
You need to compile with psram enabled in arduino. Select board ESP32 Dev Module

lljaxn, 2020-12-29:
Thank you!  I have been fighting this for a couple days.  
I tried the below changes and I am getting:
Loading config..
..............
ERROR: pSRAM must be enabled
Unable to continue, SD card fail, restart after 10 secs

This is an improvement as I was not getting as many dots before.

My current changes are
Tools->
Select Board = ESP32 Dev Module
Set Partition Scheme = Huge APP (3Meg No OTA/1Meg

Comment out the following in myConfig.h
//DNSServer dnsAPServer;
About line number 211 where "bool setWifiAP()" starts till the end TO PREVENT THE ALREADY DEFINED ERRORS

s60sc, 2020-12-30:
There is something wrong with your installation if you need to edit files. Why is the linker complaining about a rogue utils.cpp file in your sketch folder. 
The tools settings are ok

s60sc, 2020-12-30:
Try getting an example sketch for  ESP32 Dev Module working first - File / Examples / ESP32 / Camera /CameraWebserver

lljaxn, 2020-12-30:
> There is something wrong with your installation if you need to edit files. Why is the linker complaining about a rogue utils.cpp file in your sketch folder.
> The tools settings are ok

I didn't clean the folder ESP32-CAM_MJPEG2SD when I copied your ESP32-CAM_MJPEG2SD-master  :-(

lljaxn, 2020-12-30:
> Try getting an example sketch for ESP32 Dev Module working first - File / Examples / ESP32 / Camera /CameraWebserver

I can get everything but the pSRAM to work on both ESP32-CAM modules (DM and AI versions) 
I just tried a simple sketch that just calls psramInit() and it failed... I will keep looking into this.  Closing this thread for now.

closed 2020-12-30 by lljaxn