s60sc/ESP32-CAM_MJPEG2SD #21: Get fps API

quockhanh1198, 2021-02-23:
Hi, your project is very wonderful. So, I would like to view fps API by http_GET when I configured it in arduino code. Can you suggest me ? Thank you so much !

s60sc, 2021-02-23:
Thanks
I'm not sure what your question means. The fps value is returned in JSON format in response to a GET, eg {fps: "5"}

quockhanh1198, 2021-02-23:
Thanks for your feedback. More specifically, I want to GET fps api in the browser and be able to change the fps parameters with the api itself without having to adjust from the fps setting on the streaming web, eg {http://192.168.1.1/control ? var = fps & val = 10 to http://192.168.1.1/control?var=fps&val=5}.

s60sc, 2021-02-23:
You seem to have answered your own question, just type http://192.168.1.1/control?var=fps&val=5 into the browser, or send from from your own app.

quockhanh1198, 2021-02-24:
Thank you, sorry but I want to ask you 1 question please. Moreover, If I don't want to record video from SD card, I just want to stream on browser, can you suggest me, bro ?

s60sc, 2021-02-24:
if you just want to stream, use the Arduino sketch: File > Examples > ESP32 > Camera > CameraWebServer
It has the same api.

quockhanh1198, 2021-02-24:
I tried that, but it has no function to configure fps and get fps api in the browser.

closed 2021-04-01 by s60sc