s60sc/ESP32-CAM_MJPEG2SD #22: FTP Uploading only one file / doesn't upload complete folder

FReaXit, 2021-03-13:
Hi, 
when i try to ftp upload a whole folder with several files, only one file is uploaded. 
This happens either when i  upload or move. 
When i move, the file is uploaded and deleted from sd as expected, but only one file. 
When i move again, the next file is uploaded and deleted, but not all the files in the folder. 

I will upload the log from serial monitor.

FReaXit, 2021-03-13:
Unfortunately, log does'nt show anything valuable... 
I assume that the first "Uploading as AVI", followed by the dots is the file which is uploaded (it takes some time during the dots are displayed). 
The next entries "Uploading as AVI" run quite fast, it seems the the files are skipped in some way. 

Log from Serial Monitor:
_Uploading as AVI
...............
..................................................
...........................................
Processed 140 of 140 frames
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI
Uploading as AVI_

s60sc, 2021-03-14:
Select verbose mode on web page to get more logging info

closed 2021-04-01 by s60sc