s60sc/ESP32-CAM_MJPEG2SD #27: Heating & schematic

alatwifi, 2021-06-05:
Sorry iam really beginner,  can you help me and other beginners with drawing schemes for wiring (PIR & other sensor) in this sketch.

This is the best sketch I've ever tried, but I'm experiencing overheating with this sketch. especially in the orange & red part of the picture.

[image]

can only the heatsink help to cooling down or need a fan? Its temperature reaches 80°C in 20 minutes.
Did u have channel on yt for this project?
Maybe can help people like me learn more.

hope you can provide an answer. 
Thank you very much

s60sc, 2021-06-05:
what is your supply voltage

alatwifi, 2021-06-06:
> what is your supply voltage

Im use 5V 1A from powerbank

s60sc, 2021-06-06:
your voltage regulator (red) is too hot for 5V so your module may be faulty

alatwifi, 2021-06-26:
> your voltage regulator (red) is too hot for 5V so your module may be faulty

Ii used the heatsink on top of the power regulator & wifi module on the esp32cam board then i using fan 5v. 
and my problem is solved 🙏😀

btw download videos on the webserver sometimes the file can't be opened & sometimes the video doesn't download it just spins around waiting for the network. 

thank you very much for the sketch.👍

alatwifi, 2021-06-26:
1q

closed 2021-06-26 by alatwifi