s60sc/ESP32-CAM_MJPEG2SD #28: FTP server Upload

bogdanys, 2021-06-28:
I can't upload file on ftp server. I keep receiving this error: Error opening ftp connection to ftp://192.168.36.137/:21.
I don't know if I'm doing something wrong. Could you help me please?

s60sc, 2021-06-28:
There cannot be a / after the IP address

bogdanys, 2021-06-29:
It still does not work. Maybe I'm setting the wrong configuration. The configuration is as follows:
The IP address from which I access the site is: 192.168.36.137
FTP Server: ftp://192.168.36.137:21
Ftp port: 21
Ftp username: test
Ftp password: test
Ftp root dir: /

s60sc, 2021-06-29:
try connecting from windows command prompt to check ftp server

bogdanys, 2021-06-30:
I did this and it displays Connection refused.

s60sc, 2021-06-30:
then the problem is with your ftp server

bogdanys, 2021-06-30:
But is strange because with other example the ftp server works perfectly.

LeonardoZim, 2021-07-07:
I have the same problem

s60sc, 2021-07-08:
i cant fix problems with peoples ftp server connections

closed 2021-07-08 by s60sc