s60sc/ESP32-CAM_MJPEG2SD #29: WiFi kill-switch via physical button/toggle switch ?

michalrz250, 2021-07-02:
Hello,

I love your project! Been using it since August last year on 4 boards for simple home security and it's great.

I have a question - in one case one of my boards causes interference on my TV (aerial DVB-T) and I'd like to add a switch to simply turn the WiFi off when watching TV and then back on to review the recordings.

I know I could use pin 12 for the switch.
So far I was able to call WiFi.disconnect() and  WiFi.mode(WIFI_OFF);, the interference stopped. When I turned WiFI back on, it seems it did not record anything.

Is there an easy way to do this properly?

s60sc, 2021-07-13:
Thanks
Sorry, thought I'd responded to this.
Easiest would be to do a reboot via the button on the web page under Other Settings to restart wifi cleanly

michalrz250, 2021-07-16:
Okay! You mentioned simply rebooting to bring WiFi back and this nudged me in the right direction.
What I ended up doing is making a toggle switch that, when enabled, pulls pin 12 to ground.
In the main loop I poll that pin, and if low - turn off WiFi. After that, toggling the switch back to 'open' (pulled high) calls ESP's reboot function.
I had to uncomment 'checkconnection()' in the main loop so it doesn't get all panicky :)
It seems to be working alright, recordings are still being made (it seems). Oddly enough, the interference sometimes does not go away but that's the TV's fault (crap tuner).
Thank you for your response and thanks for your awesome project. Super useful and stellar - 3 cameras are on their 11th month without issues!

closed 2021-07-16 by michalrz250