s60sc/ESP32-CAM_MJPEG2SD #34: SD Card mount failed!

Shashwatraj3951, 2021-09-03:
Hello 
This project is really awesome.

But showing this issue...

E (902) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (00:00:23.605) mjped2sd: SD mount failed for 1-line mode
SD card initialization failed!!, Will restart after 10 secs
ets Jun  8 2016 00:22:57

Can you please help me to solve this issue??

s60sc, 2021-09-03:
Thanks
You need to insert a properly formatted working SD card

Shashwatraj3951, 2021-09-04:
Now working sir,
There is some fault with my sd card
Thanks for your reply...

Actually, I am running this device with the battery so I want to see the battery percentage on the web How can I do this??
Please reply sir...

s60sc, 2021-09-04:
how would you notify the battery voltage to the esp32. you could use pin 33 (the only analog available) via a voltage divider

Shashwatraj3951, 2021-09-08:
Okay sir