s60sc/ESP32-CAM_MJPEG2SD #37: fixed IP-address  not working

Ctenberge, 2021-09-22:
hi, I tried to use this project, put my credentials in myconfig.h.
Result: fixed IP-address is not taken into use, it takes it from DHCP.
Accepting that and use the camera for a while, it leaves the LAN sometimes after a few minutes but also after a few days of solid work.
My 2 questions are:
1. Why does it not take the IP-address presenting in myconfig.h? (noted "192.168.1.246")
2. Why does it goes of grid where I have taken spcial precautions to a good reception of Wifi?

s60sc, 2021-09-22:
its up to your router whether it accepts a static ip, the application cant force it.
wifi reception is handled by the esp32 firmware not the application

Ctenberge, 2021-09-23:
Hi,
my router is oké. I have 4 IP-cameras all assigned a fixed IP-address.
I have the first half of my subnet available for DHCP, the second half, up
from 128, destined for devices that prefer a fixed IP-address.
All my Raspberry Pi's are in that part and so are the cameras, FTP-server
and NAS.
Nothing wrong with my router.
And I have accepted a DHCP-address for the two ESP32-CAM devices. Both
disappeared after a while from my LAN. And I thought, maybe it's the
leasetime.
But no, The maximum lease is a week and they leave within that time. Not a
fixed time, it changes. But never for a long time.

That is why I decided to post my troubles.

closed 2021-10-02 by s60sc