s60sc/ESP32-CAM_MJPEG2SD #45: on board lamp (power-LED) is always on

Burkadius1952, 2021-11-08:
Hi ,after flashing the new release my problem is, that now the installed Lamp always is on. I can not swith it out. I do not use a microphon.
Can You help me agin ?
Best regards
Burkhard

s60sc, 2021-11-08:
My fault, use latest mic.cpp

Burkadius1952, 2021-11-08:
Ok,thanks . Will try it tomorrow.

Burkadius1952, 2021-11-09:
Hi, I have installed the new mic.cpp. But the lamp is allways on. Microphone gain is „0“.
I’m using the GPIO12 ->PIR and the GPIO03 -> DS18…
Best Regards
Burkhard

s60sc, 2021-11-10:
check startup log shows:
mic: Sound recording is Off

and check Lamp On is not selected

burkadius, 2021-11-10:
I have checked this two things_
Mic:Sound recording is off and in the browser the lamp switch is off

???

s60sc, 2021-11-10:
Try toggling on/off the lamp switch in browser. 
Try a different camera module if you have one.

burkadius, 2021-11-10:
Toggeling on/off Ihave tried, no reaction
I will try tomorow a new camera moduel, but it is also AIThinker (I think so ???) It is absolutely the same.

Von: s60sc
Gesendet: Wednesday, 10 November 2021 17:44
An: s60sc/ESP32-CAM_MJPEG2SD
Cc: burkadius; Comment
Betreff: Re: [s60sc/ESP32-CAM_MJPEG2SD] on board lamp (power-LED) is always on(Issue #45)

Try toggling on/off the lamp switch in browser.
Try a different camera module if you have one.

closed 2021-11-17 by s60sc