s60sc/ESP32-CAM_MJPEG2SD #46: Unprotected WIFI AP after Reboot

jingaworks, 2021-11-10:
All works nice and well, thank you for this.
...but, there is a problem, after I set my SSID and PASSWORD for my wireless Network, I reboot the board and... there I see a new AP in my Network manager, an OPEN unprotected network, ESP_64XXXX, It's not the initial CAM_ + ESP_getChipId().

What should I do to have the AP protected?

s60sc, 2021-11-10:
I've never come across this, so I dont have an answer. What happens if you try nd use it

jingaworks, 2021-11-10:
I've changed startWifi() in utils
I removed: 
```
if (WiFi.getMode() == WIFI_OFF) WiFi.mode(WIFI_AP);
else if (WiFi.getMode() == WIFI_AP) WiFi.mode(WIFI_AP_STA);
```
and replaced with: `WiFi.mode(WIFI_STA);`

...and in setWifiAP() I removed:
```
if (WiFi.getMode() == WIFI_OFF) WiFi.mode(WIFI_AP);
else if (WiFi.getMode() == WIFI_STA) WiFi.mode(WIFI_AP_STA);
```
and replaced with: `WiFi.mode(WIFI_STA);`

It's working. I unplugged my wifi access point, while the board connected. It starts the AP mode but still tries to reconnect...
When I plug back the Access Point, it connects back and stops that AT mode.
I hope won't affect anything else with this change. I will let you know if something goes wrong.
Is this wrong to do??

and a question.
when you make o version of this script, except the camera... well the credential part it's very nicely done, and some settings for the server to POST some data readings 👍

s60sc, 2021-11-10:
Thats fine if you just need station mode.

The credentials part was provided by [gemi254](https://github.com/gemi254)

jingaworks, 2021-11-10:
Yes, I do need only STA mode if it is connected. 
when it's not connected or gets disconnected, it starts AP mode.

jingaworks, 2021-11-10:
To answer your question "What happens if you try and use it" it works. I can see the camera interface as normal.
This is because WIFI_AP_STA mode sets the default, no password ESP AP because setWifiAP() is never called.'

jingaworks, 2021-11-10:
to check for WIFI connection I have a task that does only that, I have it from a nice tutorial I've read.

Task code:
```
#define WIFI_TIMEOUT_MS 20000
TaskHandle_t KeepWiFiAliveHandle;

// Keep wifi alive Task
void keepWiFiAliveTask(void * parameters){
  for (;;){
    if(WiFi.status() == WL_CONNECTED){
      ESP_LOGI(TAG, "WIFI still connected to: %s, IP: %s, mode: %d, status: %d, ap clients: %d", ST_SSID, (WiFi.localIP()).toString().c_str(), WiFi.getMode(), WiFi.status(), WiFi.softAPgetStationNum() );
      vTaskDelay(200000 / portTICK_PERIOD_MS);
      continue;
    }
    WiFi.mode(WIFI_AP_STA);
    ESP_LOGI(TAG, "Reconnecting to: %s, mode: %d,", ST_SSID, WiFi.getMode());
    WiFi.begin(ST_SSID, ST_Pass);

    unsigned long startAttemptTime = millis();
    while (WiFi.status() != WL_CONNECTED && millis() - startAttemptTime < WIFI_TIMEOUT_MS)  {
      Serial.print(".");
      delay(200);
      Serial.flush();
    }
    Serial.println(".");

    if (WiFi.status() != WL_CONNECTED) {
      ESP_LOGI(TAG, "WIFI Faild connecting to %s, mode: %d", ST_SSID, WiFi.getMode() ) 
      continue;
    }
    ESP_LOGI(TAG, "WIFI Reconnected to %s, mode: %d, status: %d, with IP: %s", ST_SSID, WiFi.getMode(), WiFi.status(), (WiFi.localIP()).toString().c_str() );
  }
}
```
setup code:
```
  xTaskCreatePinnedToCore(
    keepWiFiAliveTask,
    "keepWiFiAlive",
    4800,
    NULL,
    1,
    &KeepWiFiAliveHandle,
    CONFIG_ARDUINO_RUNNING_CORE);
```

maybe you can add it to the code, reboot is not an option if you want to record, it will keep rebooting and no video capturing will be done if it loses the connection.

Now, I always have AP started.  That is something to think about. 
Add input fields to be able to change the default password?. If it will be always on, then this is a security risk?
Why use WIFI_AP if it will be WIFI_AP_STA anyway?
start WIFI_AP_STA and that's it. Start softAP with credentials (defaults) and try to connect as STA if there are some stored credentials...
This task checks the connection, if it's connected, it will delay 20 seconds and starts again. If it's not connected, it will try to connect, nothing else... If no WIFI, no problem with rebooting the board and disturbing any eventual recording in progress, I guess.
I suggest you add an option to change the default password for AP mode if it will be always on, 12345678 is not a strong one 😄

jingaworks, 2021-11-10:
The AP network changes the name and I could connect to it because I previously connected to the same network but with a different name CAM_.....
This is not unprotected. security risk still remains with a default password. 

Sorry for the confusion and misunderstanding.

closed 2021-11-10 by jingaworks

nnjim, 2022-01-03:
Hi this is a great sketch, however I agree that there appears to be a security issue. 

If I install this on a new ESP, and start it up, sure enough it starts in open AP mode, where I can set the credentials of my Wifi router. Then I save and reboot the ESP and it connects to my router and camera works no problem. But the AP also starts again in open mode, and if I connect with another device to 192.168.4.1 I can drill in and see my Wifi router credentials!

I will try the changes that @jingaworks outlined above to see if that works.

Thanks

Jim

nnjim, 2022-01-04:
Hi, thank you for such a great sketch!

However I agree there is a security issue;

I found too that when I first run the ESP with this software, there is an AP started so that I can connect a device to it using password '123456789' and then set my Router credentials and save. All good so far.

But when the ESP reboots and connects to my router, an open (no password) AP also starts too.

This now means that anybody close by can connect to the ESP, then browse the settings, where they will find my router wifi credentials clearly shown!

To prevent this, I have made the changes suggeseted by @jingaworks in his post above, and that seems to have stopped the AP from starting.

Thanks and regards

Jim