s60sc/ESP32-CAM_MJPEG2SD #47: Feature Request: Rotate image 90 degrees

buteomont, 2022-01-14:
This is the only code for the ESP-Cam that I've been able to find that works correctly and doesn't have a huge lag. Well done sir!

It has a ton of features and settings, but there is one that I need that is not there - the ability to rotate the image.  The case I am using for my ESP32-cam is oriented horizontally, so that's a problem for me using this code.  Would it be too much to ask for a way to rotate the image by 90 degrees?  Alternatively, if you point me to the best place to add it in the code I could give it a go myself.
Thanks!

s60sc, 2022-01-14:
Thanks
The ov2640 does not have that facility and it would be too much work for the esp32. For downloaded avi files, viewers such as VLC have such a facility.

closed 2022-01-18 by s60sc

averter, 2025-01-31:
What about functions to mirror the video and flip it horizontally, e.g. as described [here](https://randomnerdtutorials.com/esp32-cam-ov2640-camera-settings/)? I noticed that they are called in certain locations of the code, e.g., [here](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/blob/7ea296900ab35f50c651afacd64b85a307d94285/mjpeg2sd.cpp#L981) As a dirty hack would this work? If so what needs to be added changed? Thanks

s60sc, 2025-01-31:
already implemented on web page

averter, 2025-02-01:
Thanks for your quick reply. I'd seen the capability to rotate using the "@" symbol, but if the video is streamed elsewhere, e.g., to domoticz or Homeassistant, it's not enough. I can't find the other two buttons...is it a recently added feature (I'm on v10.5.4)?

s60sc, 2025-02-01:
under Picture Settings

averter, 2025-02-01:
Thanks very much. Not sure how I missed it out, O_O, although it was 5AM when I searched for the buttons