s60sc/ESP32-CAM_MJPEG2SD #49: assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

buteomont, 2022-01-17:
After loading, this code seems to work great!  But after running the camera continually from an external wall wart for several hours (maybe a day even), it stops responding to web requests.  Connecting it to my laptop and running the serial monitor shows this error:

```
[00:06:15.189 setup] =============== Starting ===============
[00:06:15.235 prepSD_MMC] SD ready in 1-line mode 
[00:06:15.236 infoSD] SD card type SDHC, Size: 29818MB, Used space: 801MB, Total space: 29809MB
[00:06:15.437 setup] Camera init OK
[00:06:15.485 loadConfig] Loading config..
E (1174) cam_hal: EV-EOF-OVF
[00:06:15.488 loadConfig] Loaded ssid: tinyhouse pass: buteomontHill
[00:06:15.491 remote_log_init] Enabling logging mode 2

assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace:0x40083ca9:0x3ffb24a00x400905cd:0x3ffb24c0 0x4009658d:0x3ffb24e0 0x40103ea9:0x3ffb2610 0x40116c91:0x3ffb2640 0x40116cf1:0x3ffb2660 0x4010365d:0x3ffb26b0 0x400e309e:0x3ffb26d0 0x400e3732:0x3ffb2700 0x400d35eb:0x3ffb2760 0x400ec876:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0
```

The ESP32 will reboot over and over, repeating this error.
Reloading the code to the ESP32 will **_not_** fix the problem.  The only way I have found to fix it is to completely erase the ESP32 and reload the code.  In that case it will run fine for a while until the problem occurs again and I must erase and reload it again to get it to work.
It is possible that a glitch in my home router's wifi is triggering this scenario.  From my web searches, others that have experienced similar problems claim to fix it by properly initializing the wifi on the ESP32 at startup, but I have not been able to find any details on how to do this.

Thanks!

s60sc, 2022-01-17:
this line: remote_log_init] Enabling logging mode 2
indicates that it is trying to setup telnet before wifi is ready, so I need to fix this.
In the meantime, on the web page, set logging to serial and in Other Settings press Save button so that it does not try telnet on startup
PS your wifi password is shown in the log, I will delete this line in the next build.

closed 2022-01-18 by s60sc

jagonzalezga, 2022-09-05:
I've got the same problem but instead of WiFi I'm on Ethernet, i´m using websockets  and I do not understand what I have to do to make the connection to the server with the socketio library, the sketch work fine with wifi but when export the sketch to ethernet then I've got the same problem .

`/****************************************************************************************************************************
  WebServer.ino
  
  Ethernet_Generic is a library for the W5x00 Ethernet shields trying to merge the good features of
  previous Ethernet libraries
  
  Built by Khoi Hoang https://github.com/khoih-prog/Ethernet_Generic
 *****************************************************************************************************************************/

#include "defines.h"
#include <SocketIoClient.h>
#include <Arduino_JSON.h>

TaskHandle_t Task1;
SocketIoClient socketIO;
void initWebSocket()
{
  
    // server address, port and URL
    socketIO.begin("192.168.10.59",8080);
    // event handler for the event message
    socketIO.on("asignacion",getresponse);
    //socketIO.on("getPrueba",getresponse);
    delay(2000);

}

void getresponse(const char *payload, size_t length){
  Serial.printf("response: ",payload);
  String data = String(payload);
  data.replace("\\","");
  JSONVar dato = JSON.parse(data);
  Serial.println(dato["litros"]);
  }

void Task1code( void * parameter ){
  Serial.print("Task1 is running on core ");
  Serial.println(xPortGetCoreID());
  while(1){
    socketIO.loop();
    delay(100);
  }
}

void setup()
{
  Serial.begin(115200);
  while (!Serial && millis() < 5000);

  Serial.print("\nStarting WebServer on "); Serial.print(BOARD_NAME);
  Serial.print(F(" with ")); Serial.println(SHIELD_TYPE); 
  Serial.println(ETHERNET_GENERIC_VERSION);
  
#if (USING_SPI2)
  #if defined(CUR_PIN_MISO)
    ETG_LOGWARN(F("Default SPI pinout:"));
    ETG_LOGWARN1(F("MOSI:"), CUR_PIN_MOSI);
    ETG_LOGWARN1(F("MISO:"), CUR_PIN_MISO);
    ETG_LOGWARN1(F("SCK:"),  CUR_PIN_SCK);
    ETG_LOGWARN1(F("SS:"),   CUR_PIN_SS);
    ETG_LOGWARN(F("========================="));
  #endif
#else
  ETG_LOGWARN(F("Default SPI pinout:"));
  ETG_LOGWARN1(F("MOSI:"), MOSI);
  ETG_LOGWARN1(F("MISO:"), MISO);
  ETG_LOGWARN1(F("SCK:"),  SCK);
  ETG_LOGWARN1(F("SS:"),   SS);
[... 119 lines cut ...]
  {
    Serial.print(F("Speed: "));    Serial.print(Ethernet.speedReport());
    Serial.print(F(", Duplex: ")); Serial.print(Ethernet.duplexReport());
    Serial.print(F(", Link status: ")); Serial.println(Ethernet.linkReport());
  }
  Serial.println("LLego a la parte de initwebsocket y task");
  initWebSocket();
  xTaskCreatePinnedToCore(Task1code,"Task1",10000,NULL,1,&Task1,0);
  Serial.println("termino inits");

}

void loop()
{
  Serial.println("void loop");
       //socketio.loop();
       //socketIO.emit("getPrueba","{\"hello world\"}");
       socketIO.emit("setRele","{\"Dato\":\"hola\"}");
       delay(500);
       //socketIO.emit("app","{\"name\":\"hola ese\",\"chava\":\"hola chava\"}");//esta libreria solo maneja archivos json
       socketIO.emit("getPrueba","{\"name\":\"hola ese\",\"SERVIDOR\":\"HOLA MUNDO\"}");
       //socketIO.emit("setData","{\"Dato\":\"hola\",\"Dato2\":\"Andre\"}");
       //socketIO.on("get",getresponse);
        delay(500);
}`

priya2212, 2023-10-18:
Hi @jagonzalezga , have you got any solution for crashing when using ethernet? I am also facing the same error.

KingdomSon, 2024-01-21:
Any news on this? Have the same issue.

OmegaLeWolf, 2024-06-25:
Sorry for the bump, I had a similar issue with the same error code. After researching a little I found that it seemed to have been caused by the lack of initialization of the TCP/IP stack. This issue is closed with a bugfix so I'm assuming the others asking for help have a slightly different issue.
The way I solved my problem was by initializing the WiFi library and setting the modem mode to station (in my application, the other modes should work but are untested).

Hope this helps!

Kahwagi, 2024-09-24:
Hello, I get the same error using MQTT with 4G trying to receive a message with details of a binary file uploaded on an https server in order to perform FOTA, any help would be appreciated?