s60sc/ESP32-CAM_MJPEG2SD #50: Streaming fails when "show motion" is toggled and no movement occurs

buteomont, 2022-01-17:
I apologize if I'm overusing this issue facility, but I really like this software and want it to be successful!
There are actually a few (admittedly minor) issues that I'd like to see fixed:

1. If the _show motion_ switch is toggled on and then off, with no motion occurring in the FOV, then streaming stops and can't be restarted without rebooting the ESP32.
2. It apparently is not possible to stream the video to multiple destinations. I realize we are resource-constrained here but it should be possible with a reduced frame rate.  Trying to stream to multiple destinations, starting and stopping each, sometimes will hang the stream like in item #1 above.
3.  When **Logging Mode** is set to _SD card log.txt_ the log file is erased when the ESP32 is rebooted. Perhaps it needs a "Clear Log File" option instead?
4. The _Verbose_ logging switch is not saved and maintained over reboots, but is turned off automatically on reboot.

Thanks again for the only decent software available for this camera!

s60sc, 2022-01-17:
Thanks
1 has not occurred for me - were multiple streams being attempted?
2 getting a little microcontroller to service multiple streams is asking too much, a Pi based solution would be more approriate. 
3 as designed but I'll look into changing it
4 as designed, only intended for debugging

buteomont, 2022-01-17:
No, I was not messing with multiple streams at the time.  I experimented 
with this problem a bit more and the streaming was started in the main 
window. If there is any motion at all in the camera the problem does not 
occur.  Everything in the control panel works, but no streaming is 
possible until reboot.  I captured the log when it was occurring:

[21:07:21.295 remote_log_init_SD] Opened SD file for logging
[21:07:21.304 loadConfig] Setup camera_sensor, size get 36, def size: 36
[21:07:21.386 startWifi] Setting wifi hostname: espcam
[21:07:21.387 startWifi] Got stored router credentials. Connecting to: 
xxxxxxx with pass: xxxxxxx
[21:07:21.390 startWifi] Getting ip from dhcp..
[21:07:21.506 prepMjpeg] To record new MJPEG, do one of:
[21:07:21.520 prepMjpeg] - move in front of camera
[21:07:21.520 prepMjpeg]
[21:07:26.026 keepWiFiAliveTask] WIFI connected to tinyhouse, mode: 1, 
status: 3, with IP: 192.168.0.114
[21:07:26.042 setupHost] Mdns services http://espcam Started.
[21:07:26.521 setup] Use 'http://192.168.0.114' to connect to Station
[21:07:26.529 prepMic] Sound recording is Off
[21:07:26.529 OTAsetup] OTA on port 82
[21:07:26.532 setup] Camera Ready @ 20MHz, version 4.0.
[02:07:27.045 getLocalNTP] Got current time from NTP: 18/01/2022 02:07:27
[02:09:55.687 processFrame] Capture started by Motion
[02:09:56.891 isNight] Night time
[02:09:56.894 closeMjpeg] Capture time 0, min seconds: 5
[02:09:57.086 checkFreeSpace] Card free space: 28556MB
[02:09:58.181 isNight] Day time
[02:09:58.184 processFrame] Capture started by Motion
[02:10:18.434 closeMjpeg] Capture time 6, min seconds: 5
[02:10:18.645 closeMjpeg] ******** MJPEG recording stats ********
[02:10:18.647 closeMjpeg] Recorded 
/20220118/20220118_020958_SVGA_6_20_120.mjpeg
[02:10:18.648 closeMjpeg] MJPEG duration: 20.2 secs
[02:10:18.653 closeMjpeg] Number of frames: 120
[02:10:18.666 closeMjpeg] Required FPS: 20
[02:10:18.667 closeMjpeg] Actual FPS: 6.0
[02:10:18.668 closeMjpeg] File size: 3.47 MB
[02:10:18.668 closeMjpeg] Average frame length: 30294 bytes
[02:10:18.676 closeMjpeg] Average frame monitoring time: 143 ms
[02:10:18.682 closeMjpeg] Average frame buffering time: 7 ms
[02:10:18.685 closeMjpeg] Average frame storage time: 16 ms
[02:10:18.690 closeMjpeg] Average SD write speed: 1770 kB/s
[02:10:18.696 closeMjpeg] File open / completion times: 106 ms / 211 ms
[02:10:18.702 closeMjpeg] Busy: 100%
[02:10:18.708 closeMjpeg] Free heap: 133032, free pSRAM 1970247
[02:10:18.710 closeMjpeg] *************************************
[02:10:18.717 checkFreeSpace] Card free space: 28553MB
[02:11:12.105 ERROR @ mjpeg2sd.cpp:532] Not a directo

Not sure why it stopped mid-word on that last line.

s60sc, 2022-01-17:
Actual FPS: 6.0 is very low and Average frame monitoring time: 143 ms is very high. 
Did you change these values?
#define MOVE_START_CHECKS 5 
#define MOVE_STOP_SECS 2

buteomont, 2022-01-18:
> Actual FPS: 6.0 is very low and Average frame monitoring time: 143 ms is very high. Did you change these values? #define MOVE_START_CHECKS 5 #define MOVE_STOP_SECS 2

No, I don't think I changed anything other than my ssid and pw. At least not intentionally.

```
#include "myConfig.h"

// user defined environmental setup
#define MOVE_START_CHECKS 5 // checks per second for start
#define MOVE_STOP_SECS 2 // secs between each check for stop, also determines post motion time
#define MAX_FRAMES 20000 // maximum number of frames in video before auto close                                                                                                           
#define FORMAT_IF_MOUNT_FAILED true //Auto format the sd card if mount failed. Set to false to not auto format.
```

s60sc, 2022-01-26:
Just remembered that streaming is disabled if motion capture enabled to allow for motion recording until motion stops

closed 2022-01-26 by s60sc