s60sc/ESP32-CAM_MJPEG2SD #51: The lamp does not light in Night Time mode.

malpalo, 2022-01-19:
Thank you for a great project. I am a beginner and I try to try all the features of the camera. If I understood correctly, in Night Time and Motion Capture mode, the lamp will light up when there is motion. I use a PIR sensor that responds to movement and recording is activated but without the lamp on. When manually on / off the lamp works properly. Is there anything else to set? 
Thanks  Palo.

s60sc, 2022-01-19:
Thanks
Currently the lamp only works manually, as not everyone uses the lamp, or uses lamp pin for other purposes.
I will look into adding automatic lamp with PIR in the next release

malpalo, 2022-01-19:
Thanks for the explanation and I look forward to the next versions.

s60sc, 2022-01-26:
Added to next version which should be released within a couple of weeks

closed 2022-01-26 by s60sc