s60sc/ESP32-CAM_MJPEG2SD #53: the board hangs on reboot and no wifi Router available!

jingaworks, 2022-01-30:
Don't call me Mr Bug, but I like your scrip and I play with it a lot.
I discovered there is a situation the board don't start anymore.
If the board reboot and there is no router available, then the board starts the AP but never starts the server.
Because there are stored credentials, it create the keepWiFiAliveTask() and the task starts the AP and delete the task... but it never finish the setup :(
To replicate this, unplug the router and reboot the board... 
I found a way to prevent this by adding some check before it starts the keepWiFiAliveTask() and starts the keepWiFiAliveTask only if connects to WiFi router.

```
bool startWifi() {
  WiFi.persistent(false); //prevent the flash storage WiFi credentials
  WiFi.setAutoReconnect(false); //Set whether module will attempt to reconnect to an access point in case it is disconnected
  WiFi.setAutoConnect(false);
  LOG_INF("Setting wifi hostname: %s", hostName);
  WiFi.setHostname(hostName);
  if (strlen(ST_SSID) > 0) { 
    LOG_INF("Got stored router credentials. Connecting to: %s", ST_SSID);
    if (strlen(ST_ip) > 1) {
      LOG_INF("Setting config static ip :%s, %s, %s, %s", ST_ip,ST_gw,ST_sn, ST_ns1);
      IPAddress _ip, _gw, _sn, _ns1, _ns2;
      _ip.fromString(ST_ip);
      _gw.fromString(ST_gw);
      _sn.fromString(ST_sn);
      _ns1.fromString(ST_ns1);
      _ns2.fromString(ST_ns2);
      //set static ip
      WiFi.config(_ip, _gw, _sn, _ns1); // need DNS for SNTP
    } else {LOG_INF("Getting ip from dhcp..");}
    WiFi.mode(WIFI_STA);
    
    **unsigned long startAttemptTime = millis();
    while (WiFi.status() != WL_CONNECTED && millis() - startAttemptTime < WIFI_TIMEOUT_MS)  {
      Serial.print(".");
      delay(500);
      Serial.flush();
    }

    if (WiFi.status() == WL_CONNECTED) {
      create_keepWiFiAliveTask();
    } else {
      LOG_INF("Can't connected to router. Starting Access point.");
      // Start AP config portal
      return setWifiAP();
    }**

  } else {
    LOG_INF("No stored Credentials. Starting Access point.");
    // Start AP config portal
    return setWifiAP();
  }
  return true;
}
```
:) nice done, you implemented keepWiFiAliveTask()  and made the code nicer.
Keep the good work.

s60sc, 2022-01-30:
ok Mr Bug  ;)  I'll fix that in the next release - couple of weeks time

closed 2022-01-30 by s60sc

jingaworks, 2022-02-01:
@s60sc another thing.
you have a comment in the script
  /* Not working if field ST_pass type="password"
    LOG_INF("Save pass %s",ST_Pass);
  */

it does works actually, you just have to add another case (password just like text) in: 

```
        const updateValue = (el, value, updateRemote) => {
          
          if (el.id === "wifi_channel") {
            console.log("wifi_channel it's present")
            document.getElementById("esp-now").classList.remove("hidden");
          }

          updateRemote = updateRemote == null ? true : updateRemote
          let initialValue
          if (el.type === 'checkbox') {
            initialValue = el.checked
            value = value === '1' ? true : false
            el.checked = value
          } else if (el.type === 'password') {
            initialValue = el.value
            el.value = value
          } 
```

and another case (password just like text) in: 

```
function updateConfig (el) {
          let value
          switch (el.type) {
            case 'checkbox':
              value = el.checked ? 1 : 0
              break
            case 'text':
              value = el.value
              break
            case 'password':
              value = el.value
              break
```
and it works ;)

I can't make changes to the original code, i use the code (modified) for simple ESP32 boards

If you do changes for the next release... maybe you add this as well.