s60sc/ESP32-CAM_MJPEG2SD #65: Unable to turn off motion detect

pgodd, 2022-02-13:
It appears with todays build there is an issue (not sure if new) where the switch to turn off motion capture operates, but does not actually turn off motion capture.
Tested in Chrome and Safari

https://user-images.githubusercontent.com/7835223/153771025-70b33bdd-9c68-4a52-bde3-fe9476229edf.mov

[esp32CAM log.txt](https://github.com/s60sc/ESP32-CAM_MJPEG2SD/files/8056461/esp32CAM.log.txt)
.

s60sc, 2022-02-13:
do you have a pir connected, motion detect switch only applies to camera detection

pgodd, 2022-02-14:
No PIR

pgodd, 2022-02-14:
The option is being updated in the "data/configs.txt".
1. enableMotion:1 > boot > no motion detection
2. enableMotion:0 > boot > no motion detection
3. enableMotion:1 > boot > no motion detection > Enable Motion Detection switched off/on > motion detection recording begins... >Enable Motion Detection switched off > motion detection recording begins, continues recording endlessly> Enable Motion Detection switched on > motion detection recording observes set time limit
Other strange behaviour not yet understood

pgodd, 2022-02-14:
Motion detect status changes are properly logged

pgodd, 2022-02-14:
Ok. Last comment today. The option "enableMotion" variable is being updated in the "data/configs.txt" ONLY when the save button under other settings is pressed. 
Only then will motion capture will disabled on reboot.
However, motion capture can be enabled by only toggling the variable on. Motion capture cannot be stopped unless the variable is saved in "data/configs.txt" by toggling motion capture off, pressing save, and re-booting the device

KyleGoddard, 2022-02-14:
Looks like line 161 does not set a value with `enableMotion`? It just logs the passed in value.

s60sc, 2022-02-14:
Changes to settings only relate to the session by design. Press Save to make them persistent.
I'm not able to recreate your issue.

pgodd, 2022-02-14:
Ok. I think I see the problem. It seems when motion capture is turned off, no video is recorded, but the recording button in the web interface still flashes in response to motion being detected.

pgodd, 2022-02-15:
Fixed! Thanks s60sc.

closed 2022-02-15 by pgodd