s60sc/ESP32-CAM_MJPEG2SD #66: Missing compiler options

Silentmonkey, 2022-02-14:
Hi. I am using a generic Aliexpress ESP32-CAM board with an AI-Thinker ESP32-S module and a PSRAM Chip. I would like to try out this project, however, If I choose the "AI Thinker ESP32-CAM" board from the board manager I do not see either the PSRAM or Partition Scheme options. If I choose the ESP32 Wrover board I do see Partition Schemes but still no PSRAM option.

I am using Arduino IDE version 1.8.19 and version 2.02 of the Espressif Systems ESP32 board library.

Can you advise which board I should be selecting so that I may set these options as per the instalation instructions. "Compile with Partition Scheme: Minimal SPIFFS (...). and with PSRAM enabled." or are these automatically set by choosing the  "AI Thinker ESP32-CAM" board?

Many thanks
Mark

s60sc, 2022-02-14:
Select 'ESP32 Dev Module' which works for everything.

Silentmonkey, 2022-02-14:
@s60sc Many thanks for the swift response. I see both options now.

closed 2022-02-14 by Silentmonkey