s60sc/ESP32-CAM_MJPEG2SD #67: Issue with compiling sketch for ESP32-CAM

maximagination1, 2022-02-17:
Hi s60sc,

Max here from Instructables and the YouTube channel "Max Imagination".

Recently, a handful of my viewers who've been trying to recreate the ESP32-CAM WiFi Security Camera project have been complaining about their IDE giving compiling errors when trying to upload your open-source camera code to their camera boards, this now is also the case with my own camera when I reprogrammed it. 

Here's the full error message that is coming about:

"Arduino: 1.8.19 (Windows 10), Board: "ESP32 Wrover Module, Huge APP (3MB No OTA/1MB SPIFFS), QIO, 80MHz, 115200, None"

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\user_name\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi

 Not used: C:\Users\user_name\Desktop\arduino-1.8.19-windows\arduino-1.8.19\libraries\WiFi

In file included from c:\users\user_name\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\xtensa-esp32-elf\include\c++\8.4.0\bits\ios_base.h:46,

                 from c:\users\user_name\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\xtensa-esp32-elf\include\c++\8.4.0\ios:42,

                 from c:\users\user_name\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\xtensa-esp32-elf\include\c++\8.4.0\ostream:38,

                 from c:\users\user_name\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\xtensa-esp32-elf\include\c++\8.4.0\iterator:64,

                 from c:\users\user_name\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\xtensa-esp32-elf\include\c++\8.4.0\regex:43,

                 from C:\Users\user_name\Desktop\ESP32-CAM_MJPEG2SD-master-20220213T141508Z-001\ESP32-CAM_MJPEG2SD-master\ESP32-CAM_MJPEG2SD\app_httpd.cpp:21:

c:\users\user_name\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\xtensa-esp32-elf\include\c++\8.4.0\system_error:39:10: fatal error: bits/error_constants.h: No such file or directory

 #include <bits/error_constants.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board ESP32 Wrover Module."

As I'm not super skilled at the software stuff, I thought maybe you might have the solution to this recurring issue. Not only myself, but many other makers who are struggling to get their ESP cams working would appreciate any help on this from you or others from GitHub.

Thank you in advance, 
Max

s60sc, 2022-02-17:
Hi
The error lists app_httpd which is no longer part of the app. Try deleting the current sketch and instal a clean copy. Also use partition minimal spiffs if you want to use ota

s60sc, 2022-02-17:
One of the comments mentioned esp_log_system_timestamp error. This is due to using an outdated arduino-esp32 library - not to be confused with the ide version.

maximagination1, 2022-02-18:
Hi,
Thanks for this, however... I am still getting errors irregardless of the changes I made according to what you mentioned.

I updated, downgraded, and tried different versions of the esp32 library in the boards manager tab, didn't make a difference.

I am now getting this error stating that my WiFi network details are not declared in their scopes:
"Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "ESP32 Wrover Module, Huge APP (3MB No OTA/1MB SPIFFS), QIO, 80MHz, 921600, None"

In file included from C:\Users\EliteBook\Desktop\ESP32-CAM_MJPEG2SD-master\ESP32-CAM_MJPEG2SD\mic.cpp:15:0:

C:\Users\EliteBook\Desktop\ESP32-CAM_MJPEG2SD-master\ESP32-CAM_MJPEG2SD\mic.cpp: In function 'void prepMic()':

myConfig.h:294:77: error: 'esp_log_system_timestamp' was not declared in this scope

 #define INF_FORMAT(format) "[%s %s] " format "\n", esp_log_system_timestamp(), __FUNCTION__

                                                                             ^

C:\Users\EliteBook\Desktop\ESP32-CAM_MJPEG2SD-master\ESP32-CAM_MJPEG2SD\myConfig.h:295:39: note: in expansion of macro 'INF_FORMAT'

 #define LOG_INF(format, ...) logPrint(INF_FORMAT(format), ##__VA_ARGS__)

                                       ^

C:\Users\EliteBook\Desktop\ESP32-CAM_MJPEG2SD-master\ESP32-CAM_MJPEG2SD\mic.cpp:131:3: note: in expansion of macro 'LOG_INF'

   LOG_INF("Sound recording is %s", USE_MIC ? "On" : "Off");

   ^

C:\Users\EliteBook\Desktop\ESP32-CAM_MJPEG2SD-master\ESP32-CAM_MJPEG2SD\mic.cpp: In function 'void finishAudio(const char*, bool)':

myConfig.h:294:77: error: 'esp_log_system_timestamp' was not declared in this scope

 #define INF_FORMAT(format) "[%s %s] " format "\n", esp_log_system_timestamp(), __FUNCTION__

                                                                             ^

C:\Users\EliteBook\Desktop\ESP32-CAM_MJPEG2SD-master\ESP32-CAM_MJPEG2SD\myConfig.h:295:39: note: in expansion of macro 'INF_FORMAT'

 #define LOG_INF(format, ...) logPrint(INF_FORMAT(format), ##__VA_ARGS__)

                                       ^

C:\Users\EliteBook\Desktop\ESP32-CAM_MJPEG2SD-master\ESP32-CAM_MJPEG2SD\mic.cpp:166:7: note: in expansion of macro 'LOG_INF'

       LOG_INF("Captured %d audio samples with gain factor %i", totalSamples, gainFactor);

       ^

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\EliteBook\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi

exit status 1

'********hostname' was not declared in this scope

Knowing this, could you please steer me in the right direction?

Thanks

maximagination1, 2022-02-18:
Could this also be because I have multiple libraries which are clashing?
"Multiple libraries were found for "WiFi.h" "

s60sc, 2022-02-18:
The multiple libraries is not an issue - its because there are both Arduino and espressif libraries. 
The esp_log_system_timestamp error is because your arduino-esp32 version is not the latest 2.0.2.
See issue #61

maximagination1, 2022-02-21:
Hi, thanks for your current help, though, it didn't seem to work for me.
I am now using the older version of your code back from 2021, so far it compiles but with a smaller issue. It does NOT connect to my private WiFi network which I've entered my credentials for but instead chooses to start an access point that doesn't work. 
(My stored WiFi credentials have no spelling mistakes and have not been modified since).
Could you please help me out on why this is happening and what must be changed?
Thanks.

s60sc, 2022-02-21:
I'm unable to investigate older versions.
If you have a clean installation and arduino-esp32 v2.0.2 then the latest version will compile. 
Other people have raised the same issue which is always fixed by updating to the latest version.
What error are you getting.

maximagination1, 2022-02-21:
> A couple of thoughts;
> 
> When you compile/upload you need to select "ESP32 Dev Module" not 'ESP32 Wrover Module', also have PSRAM enabled, and 'Minimal Spiffs(1.9M App with OTA/190KB SPIFFS)' - because this sketch includes OTA software and may create problems when trying to compile if you don't select an OTA partion scheme?
> 
> It is known that some ESPs have a problem connecting to an access point where the SSID contains a space and/or special characters.

Thanks for the heads up on that, I've always had 'ESP32 Wrover Module' selected, never thought of this..., now I'll try uploading using "ESP32 Dev Module" instead. Okay, I'll also select that partition scheme.
Will get back to you updating the situation.

s60sc, 2022-02-21:
Only in utils.cpp. 
Only the section marked 'User modifiable defines' in myConfig.h is user modifiable
Dont modify anything that starts with the extern keyword - that a linker instruction, which is why you may be getting compilation problems.

maximagination1, 2022-02-21:
> Only in utils.cpp. Only the section marked 'User modifiable defines' in myConfig.h is user modifiable Dont modify anything that starts with the extern keyword - that a linker instruction, which is why you may be getting compilation problems.

Thanks for that, so that's why... The code now compiles every time successfully, supposedly connects with WiFi, but when I type the given IP address into my browser, nothing opens and gives me an error instead. 
In the serial monitor, it displays that "SPIFFS: mount failed" and "Not yet synced with NTP"
Why might this be?

s60sc, 2022-02-21:
spiffs isnt used. NTP is internet time.
If your IP is 192.168.4.1 then you have an AP

maximagination1, 2022-02-22:
Hi, I'm being given a WiFi IP in serial so it must mean my camera is now supposed to be connected to my private WiFi network.
Though, when I type the given IP into the browser I get "File does not exist or cannot be opened" and in the serial monitor "[12:46:56.111 ERROR @ webserver.cpp:44] File does not exist or cannot be opened: /data/MJPEG2SD.htm"

Please help...

nnjim, 2022-02-22:
Have you copied the data folder and its files onto the SD card as outlined in the readme.md?

maximagination1, 2022-02-22:
Thanks a lot for the tip @nnjim & @s60sc , I got it all working after copying those contents onto the SD card.
This really helps, I will credit you and s60sc for helping out with the situation, in the description of my YouTube video and Instructables page about my ESP32 cam CCTV camera.

Thank you so much.

s60sc, 2022-02-22:
Glad its sorted
Thanks again @nnjim

closed 2022-03-01 by s60sc

maximagination1, 2023-03-14:
> @maximagination1 <https://github.com/maximagination1>
> I was following your Youtube video and I encountered the same problem you
> posted:
> c:\users\yap chee
> kian\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\system_error:39:10:
> Error compiling for board ESP32 Dev Module.
> How did you resolve the problem?

Hi there,

This error message that's being displayed shows a library error.

It may mean you don't yet have this library installed, if so then be sure
to do so. Or this could mean that library needs updating, which you can do
in the library manage of your Arduino IDE.

P.S. Maybe even your ESP32 boards library needs updating (this can be done
in Boards Manager).

Cheers,
Max.

YapCK45, 2023-03-14:
@maximagination1 
Thanks for the reply. Updated the board to esp32 v2.0.7 and everything works fine. 

Your Youtube video for the LED variable lighting is quite bright. All my cards seem to be dimmer. Noticed that the maximum brightness is not the same as the previous for fully ON. I reckoned it is using some PWM mechanism as the light appears to be blinking.