s60sc/ESP32-CAM_MJPEG2SD #74: timelapse

smit604, 2022-03-21:
hello,
what is the right procedure to use the time-lapse mode, got accidently some xxxxT.avi files but they are all  0 mB;
Timelapse switch is on and then? 
start record?  
or autodetection of motion?
or else??

Leo Smit

smit604, 2022-03-21:
> Set time lapse on, automatically starts. File only valid when duration
> completed, then starts next one

[image: image.png]
Thank you for your quick response!!
 I set these  values (above) , do the values have to be in brackets or is
it good this way??

Leo Smit

s60sc, 2022-03-21:
image not visible
#define - brackets needed if any spaces in definition

smit604, 2022-03-21:
okay, thanks!

smit604, 2022-03-21:
*47   // timelapse - mjpeg2sd.cpp48   #define SECS_BETWEEN_FRAMES (30) //
too short interval will interfere with other activities49   #define
MINUTES_DURATION (12)         // a new file starts when previous ends50
 #define PLAYBACK_FPS (3)                  // rate to playback the time
lapse, min 1 *

Is this the correct way??

s60sc, 2022-03-21:
Yes

smit604, 2022-03-22:
Great, thanks!

closed 2022-04-07 by s60sc