s60sc/ESP32-CAM_MJPEG2SD #80: Compile error

kondorcl, 2022-04-06:
Please can someone help me.
IDE Arduino V1.8.10
ESP32 V2.0.2
ESP32 Dev Module
Partition: Minimal SPIFFS(1.9MB APP With OTA/190kB SPIFFS)
PSRAM : Enabled
ESP32-CAM_MJPEG2SD V6.2
C:\Users\xxxxxxxx\Documents\Arduino\libraries\WebServer\src/HTTP_Method.h:5:18: error: 'HTTP_GET' conflicts with a previous declaration
   HTTP_GET     = 0b00000001,

Your cooperation in this matter will be apreciated.

s60sc, 2022-04-07:
You have previously installed an Arduino webserver library that conflicts with the esp library

kondorcl, 2022-04-07:
Sir, I removed 👍 
name=WebServer
version=1.0
author=Ivan Grokhotkov

and compile again, dont have error above mentioned, but have another errors, for that I removed all libraries from C:\Users\xxxxxxx\Documents\Arduino\libraries and I have only ESP32 package libraries:
New errors found.
Several libraries were found for:
Se encontraron varias bibliotecas para "WiFiClientSecure.h"
Usado: C:\Users\xxxxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFiClientSecure
Se encontraron varias bibliotecas para "SD_MMC.h"
Usado: C:\Users\xxxxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\SD_MMC
Se encontraron varias bibliotecas para "HTTPClient.h"
Usado: C:\Users\xxxxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\HTTPClient
Se encontraron varias bibliotecas para "FS.h"
Usado: C:\Users\xxxxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\FS
Se encontraron varias bibliotecas para "SPIFFS.h"
Usado: C:\Users\xxxxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\SPIFFS
Se encontraron varias bibliotecas para "Update.h"
Usado: C:\Users\xxxxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Update
Se encontraron varias bibliotecas para "WebServer.h"
Usado: C:\Users\xxxxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WebServer
Se encontraron varias bibliotecas para "WiFi.h"
Usado: C:\Users\xxxxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi
 No usado: C:\Program
Se encontraron varias bibliotecas para "ESPmDNS.h"
Usado: C:\Users\xxxxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\ESPmDNS
Se encontraron varias bibliotecas para "Preferences.h"

s60sc thank very much for ur answer.

s60sc, 2022-04-07:
You still have multiple installations of the same libraries which you need to remove. I cant help further with arduino problems - try doing a clean install of arduino and esp board

kondorcl, 2022-04-07:
Sir, now is working.
Removed all external libraries from C:\Users\xxxxxxx\Documents\Arduino\libraries probably made a mistake earlier.. Please forget about the above comment and thank very much for help me and share ur project.

kondorcl, 2022-04-08:
s60sc, the system is working now. I had to remove all external libraries 
changing libraries name to librariesOLD and works very well Version 5, 
tomorrow I going test V6.2 with OTA web. Thank very much for ur support.

Regards,
ERamirez

s60sc, 2022-04-08:
Glad to hear it's sorted.

closed 2022-04-14 by s60sc