s60sc/ESP32-CAM_MJPEG2SD #81: ESP32-CAM V6.2

kondorcl, 2022-04-08:
s60sc, compile V6.2 no errors and doesnt works. Please could u suggest me something.

[00:00:00.898 setup] =============== Starting ===============
[00:00:01.001 infoSD] SD card type SDHC, Size: 15193MB, Used space: 49MB, of total: 15185MB
[00:00:01.197 setup] Camera init OK
[00:00:01.243 loadConfig] Load config
[00:00:01.705 WARN retrieveConfigMap] Key ST_SSID not found
[00:00:01.725 updateAppStatus] Disabling motion detection
[00:00:01.774 remote_log_init] Enabling logging mode 0
[00:00:01.785 loadConfig] Using ssid: XXXXX 
[00:00:01.785 startWifi] Setting wifi hostname: ESP32-CAM_MJPEG_7C842178
[00:00:01.785 startWifi] Got stored router credentials. Connecting to: xxxxxxx
[00:00:01.792 startWifi] Getting ip from dhcp ...
.....[00:00:04.547 startPing] Started ping monitoring 
[00:00:04.548 startWifi] Use 'http://192.168.0.11' to connect
[00:00:04.549 prepRecording] To record new AVI, do one of:

[00:00:04.565 startWebServer] Starting web server on port: 80
[00:00:04.570 startStreamServer] Starting streaming server on port: 81
[00:00:04.570 prepMic] Sound recording is Off
[00:00:04.572 prepSMTP] Email alerts not active
[00:00:04.575 checkMemory] Free: heap 118284, block: 83956, pSRAM 2564143
[00:00:04.581 setup] Camera Ready @ 20MHz, version 6.2
[21:00:04.595 WARN getLocalNTP] Not yet synced with NTP
[12:37:52.555 fileHandler] Sent /data/MJPEG2SD.htm to browser
[12:38:10.554 getLocalNTP] Got current time from NTP: 08/04/2022 12:38:10

and stays here.

kondorcl, 2022-04-09:
Sir, now is working fine. I reload all files from V6.2 data to sd data directory.

Sir, if I need battery volts in pin 33, I must unsoldier Led 1 on board?

s60sc, 2022-04-10:
Remove red led and its current limiting resistor. Ensure no more than 3.3v is applied to pin 33 from battery by adding voltage divider

closed 2022-04-14 by s60sc