s60sc/ESP32-CAM_MJPEG2SD #82: ESP32-CAM V6.2 Time Error

kondorcl, 2022-04-11:
Sir, please could you help me with wrong time display.

syncToBrowser] Sync clock to: 2022-04-11T19:14:56.791Z with tz:EST3EDT,M4.1.0/-4,M9.1.0/-3 this is ok

Camera Local Time : 2022-04-11 20:15:07 and this wrong.

Thanks,

gemi254, 2022-04-12:
Synchronizing camera time to your browser time.
Check your browser time zone and camera time zone.
What is the timezone selected in camera interface?
is this timezone: EST3EDT,M4.1.0/-4,M9.1.0/-3 correct ?

As an example..

America/Sao_Paulo                                  Zone     EST3EDT,M10.3.0/00:00:00,M2.3.0/00:00:00

"M10.3.0/00:00:00" indicates: 
M10 is 10th month (October)
.3 is the third occurrence of the day in the month (in this example, the 3rd Sunday)
.0 is Sunday 
00:00:00 is the time, (12:00 am)

gemi254, 2022-04-13:
Yes the `Time zone select` drop list is just a timezone selector to `Time zone` edit box..
Your browser/pc has a correct time / timezone ?

kondorcl, 2022-04-13:
Yes sir.
[image]
Uncomment lines 221 and 223 in utils.cpp

time_t t_of_day = mktime(&t);
  timeval epoch = {t_of_day, 0};
  struct timezone utc = {0, 0};
  settimeofday(&epoch, &utc);
  //setenv("TZ", timezone, 1);
  Serial.print(&now, "Before sync: %B %d %Y %H:%M:%S (%A) "); line 221
  getLocalTime(&now, 0);
  Serial.println(&now, "After sync: %B %d %Y %H:%M:%S (%A)"); line 223
  timeSynchronized = true;

and get this...

[21:00:04.374 WARN getLocalNTP] Not yet synced with NTP
[21:00:04.999 syncToBrowser] Sync clock to: 2022-04-13T12:24:14.869Z with tz:EST3EDT,M4.1.0/-4,M9.1.0/-3
Before sync: December 31 1969 21:00:05 (Wednesday) After sync: April 13 2022 12:24:14 (Wednesday) This is ok....
[13:24:48.363 fileHandler] Sent /data/MJPEG2SD.htm to browser
[13:24:49.136 fileHandler] Sent /data/jquery.min.js to browser

webdisplay: 1 hour early.
[image]

The recorded time stamp is fine.

[13:29:04.517 closeAvi] ******** AVI recording stats ********
[13:29:04.518 closeAvi] Recorded /20220413/20220413_132651_VGA_20_133_2658.avi

Thanks.

kondorcl, 2022-04-13:
Sir the problem was solved with this:
[image]

Thank very much..

closed 2022-04-13 by kondorcl