s60sc/ESP32-CAM_MJPEG2SD #84: Cannot connect to the IP provided (Request Time Out) on router but OK on Phone Hotspot

monroed11, 2022-04-14:
I am having an issue with accessing the Webpage. I was given IP address: [23:56:32.359 startWifi] Use 'http://192.168.2.110' to connect.

I tried it on Hotspot and it worked. My router FW is turned off. Any suggestion why it is doing this?

nnjim, 2022-04-15:
This is a problem with your network rather than the ESP, however one thought; many routers have a setting "enable isolation" or similar in the wifi settings. If this has been enabled it will prevent devices from accessing each other on the wifi.

nnjim, 2022-04-15:
And an additional thought, you do realise that an IP address beginning 192.168. is a private address - it cannot be accessed from the Internet but only locally from within the same network, unless you configure a 'port forward' on the router?

monroed11, 2022-04-16:
I am able to get the network side working. Another issue I am having is, Whenever I refresh the page from browser 3 times, I receive this error and ESP32 reboots. This is the error:
"Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core  1 register dump:
PC      : 0x4008d8e0  PS      : 0x00060730  A0      : 0x801add14  A1      : 0x3ffe8ef0  
A2      : 0x000000c7  A3      : 0x000000c3  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffc4eee  
A10     : 0x3ffc4eed  A11     : 0x3ffe90f3  A12     : 0x00000001  A13     : 0x00000136  
A14     : 0x3ffe32c4  A15     : 0x4008c864  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000c7  LBEG    : 0x4008d901  LEND    : 0x4008d911  LCOUNT  : 0xfffffff9  

Backtrace:0x4008d8dd:0x3ffe8ef00x401add11:0x3ffe8f00 0x401aa87d:0x3ffe9210 0x400d5525:0x3ffe92d0"

Cannot decode it using ESP Exception Decoder. I am receiving Decode failed error. Just want to know if you are experiencing same isssue as well.

s60sc, 2022-04-16:
to fix decode failed error see [here](https://github.com/me-no-dev/EspExceptionDecoder/issues/68#issuecomment-980978510) 

EXCVADDR: 0x000000c7 indicates bad pointer, but need decoded trace to know where.

The web page refreshes auitomatically, no need for manual refresh

closed 2022-05-04 by s60sc