s60sc/ESP32-CAM_MJPEG2SD #85: Feature Request- Password protection

rernars, 2022-04-17:
First of all thanks for creating such a great project .I am tinkering with my esp32cam and am thinking about port forwarding the stream outside of my LAN and I think a password protection on the web page would be wonderful since that would be the only( maybe except creating a vpn and connecting to the web page using that) way of accessing the stream outside of my LAN. If you have a better/more secure way of doing this i would love to hear about that!

s60sc, 2022-04-18:
thanks
i'm a c programmer, my web dev is basic, but if someone else provides the feature i'll add it.

rernars, 2022-04-18:
I'll see if i can do it myself but i have never worked with web dev inside arduino.

closed 2022-05-04 by rernars