s60sc/ESP32-CAM_MJPEG2SD #88: It is not an issue it is a question Sleep mode

llamas112, 2022-04-25:
Hello I am using only the PIR sensor for motion capture (Pin 12) and I am trying to put the camera into Deep sleep mode when a recording is saved (almost got it),  but then i want to wake up the camera using the PIR sensor whe it detectcts movement (I cant do it). ¿Is it possible? or I am wasting the time. Thanks.

llamas112, 2022-04-25:
I got it, but I can not find the best place to send the camera to sleep mode.
I've put it in several places, removing various pieces of code, so it is not the best thing.
I left the suggestion, in case anyone wants to tryit, I don't know what is the last thing that is read after finishing the recording to put the sleep there.

s60sc, 2022-04-25:
I may look into adding a deep sleep mode when I have time

s60sc, 2022-06-30:
After testing, waking up from deep sleep takes far too long

llamas112, 2022-07-01:
OK, thanks for trying it, i didn't notice that when i tested

closed 2022-07-18 by s60sc