s60sc/ESP32-CAM_MJPEG2SD #89: Email alerts not working

rernars, 2022-05-04:
While uploading the code from arduino ide i didn't switch the alerts on, but now i want to switch it on and there is no option in config to switch it on. I thought maybe if i just place my smtp info it will switch itself on but nope still it says email alerts not active. How can i switch it on?

rernars, 2022-05-04:
And I have one more question (didn't want to bother with opening another issue since its a simple question) ; I wonder what kind of files i can upload with the OTA function and what purpose do the serve? Like can i upload a new config.h ( for example to turn email alerts on) ?

s60sc, 2022-05-04:
In my config.h, set below to true and recompile
#define USE_SMTP false // whether or not to send email alerts

ota: any files used by app can be uploaded

rernars, 2022-05-04:
So, can I just upload the new config.h from OTA ? It would be really difficult for me to update the config.h since i desoldered all pins from the esp32cam

s60sc, 2022-05-04:
Yes

rernars, 2022-05-06:
well, it didnt work. 
[image]

closed 2022-05-21 by s60sc

rernars, 2022-07-11:
[image]
Well it didn't work as well , sorry for late reply