s60sc/ESP32-CAM_MJPEG2SD #96: feature request

SETHSETH, 2022-06-04:
- Save the JPEG on SD Card and Upload on FTP as option when AVI not needed (mabe 5 JPEG every second for the next 5 seconds, when motion is triggert)
- customizable filename
- PHP-Upload as option
- Pan Tilt Servo 
- capability

Then it will be better than any Firmware on CCTV Cameras ;-)
I personaly wold pay for such a sketch.

s60sc, 2022-06-04:
Its not something I have time to do, but anyone else is welcome:
- there are online tools to convert avi to jepgs
- this (old) fork includes pan/tilt capability [ESP32-CAM_MJPEG2SD-PanTiltServo](https://github.com/Styne13/ESP32-CAM_MJPEG2SD-PanTiltServo)

s60sc, 2022-06-04:
I've added the pan/tilt code from the fork but I'm unable to test it as I dont have a pan/tilt device

closed 2022-06-19 by s60sc