s60sc/ESP32-CAM_MJPEG2SD #97: Can't connect to upload code.

C4PT41ND34DP00L, 2022-06-05:
I am using the Arduino IDE. ESP32 V2.02. A ESP32-Cam-mb to program. My Cable is good. Running Windows 11
When I try and upload the code I get the below

esptool.py v3.1
Serial port COM6
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

I have one I programed awhile back but I dnt remember how to get the the OTA page on that one.
I tried the below settings in IDE
Board: AI Thinker ESP32 Cam
CPU Freq: 240Mhz (wifi/bt)
Flash Freq: 80Mhz and 40Mhz
Flash Mode: QIO

Board: ESP32 Cam Wrover Module
Upload Speed: 115200
Flash Freq: 80Mhz and 40Mhz
Flash Mode: QIO
Partion Scheme: Minimal SPIFFS(1.9MB APP with OTA/190KB SPIFFS)

s60sc, 2022-06-05:
esp-cam flash mode can be flaky with some sd cards - try removing the card prior to upload

C4PT41ND34DP00L, 2022-06-05:
I did try that as well still no luck.

C4PT41ND34DP00L, 2022-06-05:
I got it to work on a windows 10 machine and will try on my windows 11 next. Had to hold the IO button while plugging it in

C4PT41ND34DP00L, 2022-06-06:
OK to update using the esp32-cam-mb Hold down the IO (boot) button while plugging in the device. This it will upload.

closed 2022-06-06 by C4PT41ND34DP00L