s60sc/ESP32-CAM_MJPEG2SD #103: Cannot get the interface to work and there is no stream.

martinmutter, 2022-06-11:
It streams but only when I do 192.168.x.xxx:81/Stream

Files on SD card and serial monitor also confirms it works.

But nothing happens when I press anything on the interface. The buttons flash green and then turns red again. And the serial monitor doesn't indicate anything happening. The only time it does anything in monitor is when I cover the camera and causes it to go into nightmode (which is how I found out the camera even worked). 

Reinstalled it from scratch 10 times. The example ESP32 cam works great. But lacks all the nice features yours has so reallyreally want to get this working.

The only things I have changed/added is SSID and wifi password. That's it.

I have  no idea what I'm doing wrong here. Followed the guide [Max Imagination](https://www.instructables.com/member/Max+Imagination/)

Could it be an knockoff brand ESP32? Would that affect the webservers ability to communicate with the Camera?

[image]
[image]
[image]

s60sc, 2022-06-11:
No idea, but not a hardware problem.
Try a different browser.
In Chrome or Edge press Ctrl-Shift-I (letter i) together to bring up inspection screen and see what's logged when you interact with the page

martinmutter, 2022-06-12:
This is what I can find when inspecting. I am not sure what you want me to look for but this is what I found relating to the stream button and the only thing the console seemed to care about.

Not a programmer (beyond copying and pasting until it "works" together) so I am not sure what I need or can provide you with to maybe help me with this. Using Brave (PC and Android) and have tried in chrome and edge as well. Will continue trying in a bunch of other browsers. 

[image]
[image]

martinmutter, 2022-06-12:
Firefox threw me this

[image]

s60sc, 2022-06-12:
The page doesnt work because of the error. Do you have a valid jquery.min.js in the data folder on your sd card? Try loading a fresh copy from the repository.

martinmutter, 2022-06-12:
Well took a look and yea. The script was 0kb. So directly pasted the one from the zip. But didn't change anything. Might have to look into a different way to install? 

Currently launching the .ino in the folder with -master removed. Adding wifi stuff. Then upload. Without the SD in. (since the guide said it should be put in after installing)

Could the issue be the 32gb sd card? It's a Sandisk Extreme HC I. So it's a rather good SD card but I know some things have issues with the HC cards.... Just spitballing longshots here.

nnjim, 2022-06-13:
Note that Expressif state the esp32-cam officially supports a maximum size of 4GB microSD card. Many have got larger sizes to work, but I've only managed to get a max of 16GB to work. Note also that it must be formatted in fat32 format.

s60sc, 2022-06-13:
I've used 32gb card ok. The problem is related to your installation, which I cant help further with.

martinmutter, 2022-06-15:
Well just gonna have to reinstall untill it works I guess. It might be an offbrand ESP32 CAM module. So might have to figure out which board I have to pick for installation....

closed 2022-06-19 by s60sc