s60sc/ESP32-CAM_MJPEG2SD #105: Included WatchDog Timer interrupt

YapCK45, 2022-06-19:
This is not an issue. I have implemented the code but not familiar with github to include the codes below. The system would restart if the normal execution of the code somehow get delayed. It would be good it is checked and verified independently.

Insert the code below before the configure camera code:

**//Watchdog set to boot if not reset within 4 seconds (4000000uS)
//Cannot be too short so that WiFi can be established. 4 - 5 secs is about right
  watchdogTimer = timerBegin(0,80,true);  //timer 0, divisor 80
  timerAlarmWrite(watchdogTimer, 4000000, false); //set time in uS must be fed within this time or reboot
  timerAttachInterrupt(watchdogTimer, &interruptReboot, true);
  timerAlarmEnable(watchdogTimer);  //enable interupt**

In the void loop include:

**void loop() {
//Reset watchdog every loop, otherwise restart videocam
  timerWrite(watchdogTimer,0);  //reset timer (feed watchdog :-)
}**

Lastly provide the interrupt routine (not sure whether the LOG_ERR would execute; not getting it in the LOG):

**void interruptReboot(){
  Serial.println("Rebooting...\n\n");
  LOG_ERR("ESP32Cam Rebooted"); 
  esp_restart();
}**

s60sc, 2022-06-19:
Not sure what this wants to achieve.
I deliberately avoid delays by using tasks, which is why loop() is empty
There is a built in [Task Watchdog Timer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html)

YapCK45, 2022-06-19:
Wow, the ESP32 is truly versatile. 
There is a manual reboot. I the watchdog timer interrupt basically to monitor that the processes are running properly, other it would restart everything. What I included basically does the same thing as the built in Task Watchdog Timer albeit more cumbersome. I agree using the built in one would be more efficient. 
Please close the remarks on Watchdog Timer.

I also tested the Pan Tilt code on a PT platform and it works with some modifications. Here are the main points:

1. Pin 4 cannot be used as it is tied to the LED. Instead, use pin 13 for Pan. Pin 12 for Tilt is okay.
2. The numbers (0 to 180) for Pan and Tilt is intuitively wrong - 0 corresponds to right most and 180 corresponds to left most. Use 180 - panVal and it corresponds with the movement. I don't reckon it has anything to do with the SG90s. Check it out anyway.

Since so much effort had be spent on improving the efficiency and optimizing the code, would you consider using core 2 to divide out the  tasks. This should improve the performance. It's a good challenge though.

s60sc, 2022-06-19:
The ESP automatically assigns tasks to cores.
Thanks for testing pan/tilt - I'll look into it further when I receive servos

C4PT41ND34DP00L, 2022-06-22:
> I also tested the Pan Tilt code on a PT platform and it works with some modifications. Here are the main points:
>
> 1. Pin 4 cannot be used as it is tied to the LED. Instead, use pin 13 for Pan. Pin [...]

I was wondering if you got any errors when compiling with the Pan Tilt added?
I wanted to check it out and get an error no matching function for call to 'Servo::attach(int, int, int, int, int, int)'
what version of ESP32Servo library did you use?

YapCK45, 2022-06-23:
You are using the wrong servo library. Refer to the include <Servo.h> in panTilt.cpp section in the sketch.

C4PT41ND34DP00L, 2022-06-23:
> You are using the wrong servo library. Refer to the include <Servo.h> in panTilt.cpp section in the sketch.

That gives an error for programing the ESP32 cam AI thinker

YapCK45, 2022-06-23:
You can use "ESP32 Dev Module" with partition scheme "Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)", enable PSRAM, and select your COM port as in the [instructable](https://www.instructables.com/How-to-Make-a-WiFi-Security-Camera-ESP32-CAM-DIY-R/) by [Max Imagination](https://www.instructables.com/member/Max+Imagination/).
I don't have a problem using ESP32 cam AI Thinker.

closed 2022-07-18 by s60sc