s60sc/ESP32-CAM_MJPEG2SD #107: Problem programming ESP32-CAM_MJPEG2SD

YapCK45, 2022-06-23:
Not sure what is causing this problem. I have been using the ESP32-CAM_MJPEG2SD code over the last few days. When I flashed it on a new ESP32Cam board and after it rebooted, I got the following:

18:52:49.643 -> rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
18:52:49.643 -> configsip: 0, SPIWP:0xee
18:52:49.643 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:52:49.643 -> mode:DIO, clock div:1
18:52:49.643 -> load:0x3fff0030,len:1420
18:52:49.643 -> ho 0 tail 12 room 4
18:52:49.643 -> load:0x40078000,len:13540
18:52:49.643 -> load:0x40080400,len:3604
18:52:49.643 -> entry 0x400805f0

I verified that it on a working ESP32Cam module with the same results. The puzzling thing is that I could flash it with some other code without problem. I am not sure if this is hardware or software related. Appreciate any comments or help.

nnjim, 2022-06-23:
Results of Googling this error seems to suggest that you may have a power issue. This can possibly be resolved by having a better power supply or  lead and/or putting a large(r) electrolytic capacitor 100uF (or more) across the power pins on the ESP.

YapCK45, 2022-06-23:
Power supply is a possibility. But it worked earlier and also with other codes and I had also disabled brownout detector.
I finally selected ESP Cam AI Thinker and it worked as before. This is mysterious.

closed 2022-06-23 by YapCK45