s60sc/ESP32-CAM_MJPEG2SD #112: Initialize servo to mid-position and to provide for clock-wise or counter clock-wise rotation

YapCK45, 2022-07-30:
Congratulations on version 7 of the software. There is much improvements over v6.2.3 in particular changing configuration over the browser. Two issues noted with reference to the servos:
a. Initialize the servo to position 90 for both pan and tilt on start up. Noticed it is not powered until some changes in the position albeit may not be used. I think it would be good to have it in a known state. Currently, they are undefined.
b. Provide for clock-wise or counter clock-wise motion. Currently I have to use 180-angle in line 128 of peripherals.cpp for my servo to work. Again this can be configured over the browser.

Just two noteworthy comments and one question:
a. It is really good that the lamp can be configured over the browser - both the pin used and whether it can be turned on/off
b. Timelapse value also set in the configuration
c. What is the purpose of the % bar below select file/folder

Red is quite standard for the buttons but would like to request that the button color can be configurable.

Still exploring the rest of the changes. Great job!

s60sc, 2022-07-30:
Thanks

> a. Initialize the servo to position 90 for both pan and tilt on start up. Noticed it is not powered until some changes in the position albeit may not be used. I think it would be good to have it in a known state. Currently, they are undefined.

Fix uploaded

> b. Provide for clock-wise or counter clock-wise motion. Currently I have to use 180-angle in line 128 of peripherals.cpp for my servo to work. Again this can be configured over the browser.

Can switch min max angles in config

> c. What is the purpose of the % bar below select file/folder

FTP upload progress

> Red is quite standard for the buttons but would like to request that the button color can be configurable.

change to MJPEG2SD.htm uploaded, edit:
```
        <style>
            :root {
            --itemColor: red;
            }
```

YapCK45, 2022-07-31:
Thanks for the update. I have closed the issue.

closed 2022-07-31 by YapCK45