s60sc/ESP32-CAM_MJPEG2SD #113: Timelapse triggered a reboot

YapCK45, 2022-07-31:
Tried out the time lapse using:
a. Time lapse duration = 30 minutes
b. Time lapse playback FPS = 1
c. Time lapse interval (secs) = 15

This should capture 120 frames. The serial monitor on the development kit shows that when the time lapse button was clicked, it triggered a reboot. Screen shot below:
[image]

s60sc, 2022-07-31:
might be insufficient ram, I'll investigate

YapCK45, 2022-08-01:
I think that is probably correct. 
I disabled motion detect and it worked without triggering the reboot. 
Apparently some other events also be triggering that and sometimes the serial monitor stopped capturing. I had to restart another instance for it to capture properly.
Saw something about IO Extender not used when the system booted up and peripherals being hosted on separate ESP in the update documentation. Not sure what that is about.

s60sc, 2022-08-01:
updated to reduce ram usage

YapCK45, 2022-08-05:
Version 7.2 does not seem to work. Enabling time lapse still triggering a reboot. Double checked it on 2 ESP32-Cam modules and also using lower resolution.
Please confirm that the partition scheme to use.

YapCK45, 2022-08-13:
Done some more testing. Seems like Enable Motion Detect and Save Capture can be dependent on the hardware. Checked on 2 ESP32-Cam with the came code (v7.2). One work without problem while the other would trigger a reboot when motion is detected. Information captured on serial monitor:
[image]

Any suggestion how to mitigate this issue?

closed 2022-08-24 by s60sc