s60sc/ESP32-CAM_MJPEG2SD #117: Motion Detection and Save Capture is hardware dependent

YapCK45, 2022-08-13:
Tested the Motion Detect and Save Capture function. Seems like they could be dependent on the hardware. Checked on 2 ESP32-Cams with the came code (v7.2). One works without problem while the other would trigger a reboot when motion is detected. Information from serial montior:

[image]

Motion will consistently trigger a reboot. I did not realize this until I saw it on the serial monitor when checking another problem. Probably won't notice this at all and would be wondering why there is no capture. And it is on only **one** of two ESP32-Cam modules. Any idea what is causing it and how to mitigate this issue?

s60sc, 2022-08-13:
You need to decode the stack trace: https://github.com/me-no-dev/EspExceptionDecoder

YapCK45, 2022-08-13:
Tested the Motion Detect and Save Capture function. Seems like they could be dependent on the hardware. Checked on 2 ESP32-Cams with the came code (v7.2). ~~One works without problem while the other would trigger a reboot when motion is detected. Information from serial montior:~~ Rechecked again and it seems both have the same problem. So it is unlikely to be the hardware. Checked the same hardware with v6.2.3 and have the same problem. The puzzling part is that I have one unit running v6.2.3 for days without problem.
Decoding the stack trace is another level of complexity. Not sure I want to go there

YapCK45, 2022-08-13:
I think you can close this.

closed 2022-08-13 by s60sc