s60sc/ESP32-CAM_MJPEG2SD #118: Ver 7.2 - Resolution setting keeps reverting to SVGA(800x600)

YapCK45, 2022-08-13:
Save Capture keeps reverting to SVGA(800x600) when a higher resolution is selected. I can see the log 'Sent /data/MJPEG2SD.htm to browser'. The captured video showed that SVGA is recorded. Resolution lower than this is not affected.

s60sc, 2022-08-13:
There may be something wrong with your installation. I'm seeing none of the problems you are reporting and no one else has reported them

YapCK45, 2022-08-14:
Thank you for your patience. I will try installing from scratch. I need to know:

1. The esp32 board manager to be used. I am using v2.02 as I get compilation error using v2.03 and the latest v2.04.
2.  Use ESP32 Dev Module for Board

Please close this issue after your reply. Thank you once again.

s60sc, 2022-08-14:
1 You should not get compilation issues with 2.0.4. 
2 is correct
Also load the latest contents of /data

closed 2022-08-14 by s60sc