s60sc/ESP32-CAM_MJPEG2SD #119: i get this error while compailing

ryzencraft, 2022-08-19:
c:/users/42077/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/ 8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\peripherals.cpp.o: in function `prepDS18B20()':
peripherals.cpp:(.text._Z11prepDS18B20v+0x0): multiple definition of `prepDS18B20()'; sketch\ds18b20.cpp.o:C:\Users\42077\Documents\Arduino\ESP32-CAM_MJPEG2SD/ds18b20.cpp:63: first defined here
c:/users/42077/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/ 8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\peripherals.cpp.o: in function `readDS18B20temp(bool)':
C:\Users\42077\Documents\Arduino\ESP32-CAM_MJPEG2SD/peripherals.cpp:230: multiple definitions of `readDS18B20temp(bool)'; sketch\ds18b20.cpp.o:C:\Users\42077\Documents\Arduino\ESP32-CAM_MJPEG2SD/ds18b20.cpp:80: first defined here
collect2.exe: error: ld returned 1 exit status
Multiple libraries "WiFi.h" were found
 Used: C:\Users\42077\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\WiFi
 Unused: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
A compilation error occurred with the AI ​​Thinker ESP32-CAM board.

s60sc, 2022-08-19:
Do a clean install

ryzencraft, 2022-08-19:
ok

ryzencraft, 2022-08-21:
i still got this error:
c:/users/42077/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\42077\AppData\Local\Temp\arduino-sketch-0DDB67338B8413E534BD9B88B9E8393C\sketch\peripherals.cpp.o: in function `prepDS18B20()':
peripherals.cpp:(.text._Z11prepDS18B20v+0x0): multiple definition of `prepDS18B20()'; C:\Users\42077\AppData\Local\Temp\arduino-sketch-0DDB67338B8413E534BD9B88B9E8393C\sketch\ds18b20.cpp.o:C:\Users\42077\Downloads\ESP32-CAM_MJPEG2SD-master\ESP32-CAM_MJPEG2SD/ds18b20.cpp:63: first defined here
c:/users/42077/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\42077\AppData\Local\Temp\arduino-sketch-0DDB67338B8413E534BD9B88B9E8393C\sketch\peripherals.cpp.o: in function `readDS18B20temp(bool)':
C:\Users\42077\Downloads\ESP32-CAM_MJPEG2SD-master\ESP32-CAM_MJPEG2SD/peripherals.cpp:230: multiple definition of `readDS18B20temp(bool)'; C:\Users\42077\AppData\Local\Temp\arduino-sketch-0DDB67338B8413E534BD9B88B9E8393C\sketch\ds18b20.cpp.o:C:\Users\42077\Downloads\ESP32-CAM_MJPEG2SD-master\ESP32-CAM_MJPEG2SD/ds18b20.cpp:80: first defined here
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

s60sc, 2022-08-21:
You have still not done a clean install, there is no ds18b20.cpp in the latest version.

closed 2022-08-21 by s60sc