s60sc/ESP32-CAM_MJPEG2SD #124: HARestAPI

tripunto, 2022-09-03:
hello again!
I am making progress implementing the HARestAPI library, but I am stuck, I don't know where to put the code to store the token received from home assist. I could add a long-term token in the configuration, but I find it much more attractive to use the link created by the sketch to get the token when I log in and be redirected back to the mjpeg2sc website with a topo link: http://ip_esp32cam/ control?savecode=*****token*****

I know that I'm very green in programming and I know that my code can be improved, but I think that an integration similar to the sketch would do very well.

I still have to investigate how to tell HASSIO the available sensors and how to change the parameters, but at least I have discovered that I can use http://ip_esp32cam/control?enableMotion=1 to activate the sensor or use 0 to deactivate it.

I have added these lines in the configs.txt file

ha_debug::99:na
ha_enabled::99:na
ha_port::na
ha_server::99:na
ha_ssl::99:na
ha_token::99:na

and this code in the .htm file

						  <h3>Home Assistant settings</h3>
						  <div class="input-group">
                              <label for="ha_enabled">HASSIO Enabled</label>
                              <div class="switch">
                                  <input id="ha_enabled" type="checkbox" value="0" class="default-action">
                                  <label title="Enable/disable Home Assistant Integration" class="slider" for="ha_enabled"></label>
                              </div>
							  <div class="switch" align="right" width="150px" id="newToken" style="display:block">
							  	&nbsp;&nbsp;<button title="Connect with Home Assistant to get an access token for this device" id="getToken" style="top:-32px" onClick="getToken()">Get Token</button>
							  </div>
                          </div>

						  <div class="input-group" id="hassio-group">
                              <label for "ha_server">HASSIO Server</label>
                              <input id="ha_server" name="ha_server" length=32 placeholder="Home Assistant Server Host" class="default-action">
						  </div>
                          <div class="input-group" id="hassio-group">
                              <label for "ha_port">HASSIO Port</label>
                              <input id="ha_port" name="ha_port" length=6 placeholder="Home Assistant Server Port" class="default-action">
                          </div>
                          <div class="input-group">
                              <label for="ha_ssl">HASSIO SSL</label>
                              <div class="switch">
                                  <input id="ha_ssl" type="checkbox" class="default-action" value="0">
                                  <label title="Enable/disable Home Assistant Server SSL" class="slider" for="ha_ssl"></label>
                              </div>
                          </div>
						  <div class="input-group">
                              <label for="ha_debug">HASSIO Debug</label>
                              <div class="switch">
                                  <input id="ha_debug" type="checkbox" class="default-action" value="0">
                                  <label title="Enable/disable Home Assistant Debug" class="slider" for="ha_debug"></label>
                              </div>
                          </div>
                          <div class="input-group" id="hassio-group">
                              <label for "ha_token">HASSIO Token</label>
                              <input id="ha_token" name="ha_token" length=32 placeholder="Home Assistant Active Token" class="default-action">
						  </div>
						  <div class="input-group" id="hassio-group">
                              <label for "ha_pwd">HASSIO tmpToken</label>
                              <input id="ha_pwd" name="ha_pwd" length=32 placeholder="Home Assistant Udate Token" class="default-action" disabled>
						  </div>
                          <br>
		<script>
			function getToken() {
				var hostURL = window.location.href;
				var hostName = document.getElementById("hostName").value;
				var haServer = document.getElementById("ha_server").value;
				var haPort = document.getElementById("ha_port").value;
				var haSSL = document.getElementById("ha_ssl").value;
				if(haSSL == "1"){ var haProtocol = "https://"; } else { var haProtocol="http://"; }
				var getTokenURL = haProtocol + haServer + ":" + haPort + "/auth/authorize?client_id=" + hostURL + "&redirect_uri=" + hostURL + "control?&state=http://" + hostName;
				alert("GetTokenURL: "+getTokenURL);
				window.location.href = getTokenURL;

			}

		</script>

s60sc, 2022-09-04:
I've not used HA so cant comment, but if you get it working I can incorporate.
You can control all the settings using the /control?setting=value method
Best way to learn code is to try it - i am self taught

tripunto, 2022-09-04:
It is true, the best way to learn is trying to do it and I will try to do it as best as possible.

I integrated the camera in the home assistant with an addon for cameras of this type, but it did not allow me to configure automation, just see a static image or activate streaming, there was not even a need to activate a long-term token.

with the HARestAPI library, it is necessary to obtain a long-term token from the user profile, within home assistant and I have managed to obtain the token from the .htm of the camera

I have also managed to send him requests from the camera, but the responses from HA have not been good so far.

I'll keep you informed... I'm sure a lot of people will find it interesting to see this integration with home assistant in the sketch

tripunto, 2022-09-05:
dos de traducción
It is costing me more work than expected, but now I have managed to save the token

home assistant returned me ?control?token&code=***token***

in the end, I have made it return me to the .htm page and I have added the necessary javascript to read and separate the data from the url and then load an iframe with the url that adds the obtained token and then sends me back to the page .htm

I can start working on the topic of sensors!!!!!

tripunto, 2022-09-09:
s60sd, i need to get a string from Serial.read or Serial.readString so I don't have to upload the sketch every time

I need to test strings that will be sent to home assistant.

The problem is that the viod loop() doesn't repeat and I don't know how I can make this work:

  if (Serial.available() > 0) {
    String serialString = Serial.readString();
    Serial.print("I received: ");
    Serial.println(serialString);
  }

s60sc, 2022-09-09:
comment out` vTaskDelete(NULL);` to reinstate loop task

tripunto, 2022-09-13:
I can't save long token in config.txt file, it's too long, 183 characters

esp32 resets when entering a 183 character token
I have tried loading: char configLine[100]; to 250 and stops rebooting, but doesn't save the token

I had to define it in the sketch and it cannot be edited once compiled

I intend to be able to obtain a token from the camera page and have the option to use the long-term token

s60sc, 2022-09-13:
length needs to be changed in different places.
to make it simpler, an updated` prefs.cpp `has been uploaded
change` FILE_NAME_LEN` in `globals.h` to required value

tripunto, 2022-10-02:
It's getting a bit complicated for me, but I'm still at it... the home assistant documentation seems to be not entirely correct or me and the google translator don't get along.

I'm moving forward, but in very small steps

tripunto, 2022-10-10:
I have managed to register the device in Home Assistant from the .htm page of the micro-sd, after identifying myself and obtaining the authorization code, necessary to obtain the refresh token and the access token, but I still need to learn how to integrate the entities to be able to control the sensors and switches.

I'm trying not to edit anything in your .cpp and .h files, but I don't like the idea of ​​opening another port to receive RestAPI data, so I've added some lines on the web server to collect the returned data. by hassio in the path http://hostName.local/rest?

I could avoid adding those lines by opening the same port in esp32cam as the one defined by the user to connect to the home assistant (default: 8123), but it seems a bit absurd to me to open it, when it is enough to add 2 or 3 lines in the server that it already runs on port :80.

by the way... do you know any function that extracts the content of a json and converts it into variables?
For now I have had to use the JsonStreamingParser2 library, but it seems like too much code.

I'm looking to create a function, but it makes me lose a lot of time, which I could spend on the subject of entities.

I'll keep you informed!

s60sc, 2022-10-10:
updateHandler() in webserver.cpp extracts non hierachical json

tripunto, 2022-10-10:
Thanks, I'll look at it tomorrow... it's already late

tripunto, 2022-10-24:
Hello again!
I'm still moving forward with the topic, I've managed to register the device and I'm starting to register the sensors, although I still don't know very well how to declare each type of sensor, mainly the motion sensor, the light sensor and the LED flash, so that they work correctly

on the other hand... is it necessary that certain functions of the webServer.cpp be static?

I don't know how I can call them since they are declared like this... I have copied indexHandler, fileHandler and sendChunks in my .cpp, so that when a query is not valid it loads the original page inside /rest

It would be good if you declared them as global

s60sc, 2022-10-25:
remove static flag, declare in globals.h 
Didnt think HA was that complicated.
At some point I will look at using Telegram for remote control

tripunto, 2022-10-25:
That might work, but I try not to touch anything in your project.
I want to publish a single .cpp file and for it to work, it only needs two lines in: void startWebServer() {}

#ifdef HASSIO
httpd_uri_t restUri = {.uri = "/rest", .method = HTTP_GET, .handler = RestAPI, .user_ctx = NULL};
#endif

#ifdef HASSIO
httpd_register_uri_handler(httpServer, &restUri);
#endif

then only the global variables would be missing and add the code of bool updateStatus(){}

closed 2023-04-16 by s60sc