s60sc/ESP32-CAM_MJPEG2SD #125: Library or file not exist

Abasithdev, 2022-09-04:
hi guys

I found error when compile the script. the error refer to file/library not found. here is list of that library

in globals.h
`//#include <driver/i2s.h>
//#include "esp_http_server.h"
//#include <ESPmDNS.h> 
//#include "lwip/sockets.h"
//#include <vector>
//#include "ping/ping_sock.h"
//#include <Preferences.h>
//#include <regex>
//#include <SD_MMC.h>
//#include <SPIFFS.h>
//#include <sstream>
//#include <Update.h>
//#include <WebServer.h>
//#include <WiFiClientSecure.h>
//#include "esp_camera.h"`

in peripherals.cpp
`//#include "driver/ledc.h"`

in setupAssist.cpp
`//#include <HTTPClient.h>`

in utils.cpp
`//#include <esp_wifi.h>`

in uart.cpp
`//#include "driver/uart.h"`

do you know the solution?

s60sc, 2022-09-04:
Something wrong with your arduino installation

tripunto, 2022-09-04:
I had problems the other day and it turned out that I had the arduino configured to show all the compilation and upload details

I found 2 ways to solve it, configure arduino to only show debug information when compiling or add a line to globals.h file

#pragma GCC diagnostic ignored "-W***

closed 2022-09-07 by s60sc