s60sc/ESP32-CAM_MJPEG2SD #127: FTP Transfer - all files 0KB

akadow, 2022-09-06:
Hi All

I am finding an issue when auto uploading via FTP after a capture is started by motion; error as below:

17:09:33.751 -> [06:08:26.750 ftpStoreFile] Upload file: 20220906_060821_HVGA_23_5_120.avi, size: 0.6MB
17:09:33.798 -> [06:08:26.813 ERROR @ ftp.cpp:54] Command STOR  got wrong response:  Data connection already open; Transfer starting.

This results in a succsessful save of a file via FTP but each file on the FTP server is found to be 0KB in size. The corresponding file on the SD card is fine. 

The FTP server is working well, is in use for other purposes (WIN SRV 2022) and has been checked OK using other methods at the time of failure.

I note that this issue had been raised outside of Github but I cannot find any solutions.

Thanks to all for your kind help and for maintaining such an awesome project.

Cheers
Alex

s60sc, 2022-09-06:
Thanks
The FTP server may be returning a 125 when the app expects 150
As a hack, in `ftp.cpp` change line 136 from:
`  if (!sendFtpCommand("STOR ", ftpSaveName, "150")) return false;`
to
`  if (!sendFtpCommand("STOR ", ftpSaveName, "999")) return false;`

akad90, 2022-09-06:
Hi s60sc,

Brilliant solution - all fixed!

Have spent a few hours on this one so far.

Thanks for your help and for the quick reply.

Cheers
Alex

closed 2022-09-07 by s60sc