s60sc/ESP32-CAM_MJPEG2SD #132: Playback not working in ver 7.4

YapCK45, 2022-09-26:
The new GUI is quite nice. But there are two problems:

**The playback button causes a reboot**
After the avi file is selected, clicking Start Playback triggers a reboot. This is detected if the ESPCAM32 is connected to the serial monitor.
[image]

**Maximum resolution is SVGA(800x600)**
Resolution is set to SVGA(800x600) on power up. Setting the resolution to anything higher triggers a reboot and core dump. Again captured on serial monitor.
[image]

s60sc, 2022-09-26:
Cant get above issues to occur and v7.4 was only UI changes. Increased task sizes in case but unlikely to be issue. Wdt timeout indicates somethhing is causing the esp to hang, but without decoded stack traces I cannot investigate further.

closed 2022-09-26 by s60sc

YapCK45, 2022-09-27:
I can find out how to decode the stack traces and post them. Don't want to take up too much of your time.
Noticed that the wdt timeout were not CPU0 and CPU1 although the compilation selected **CPU1** only for event and arduino. Would it make any if both were selected?

s60sc, 2022-09-27:
Installing and using decoder is straightforward - [see here](https://github.com/me-no-dev/EspExceptionDecoder)

YapCK45, 2022-09-28:
I have installed the ESP Exception Decoder tool and captured the stack traces. It turns out there are two traces - one for CPU 0 and one for CPU 1 - for each of the issues.

**Exception due to playback**
CPU 0
[image]
CPU 1
[image]

**Exception due to changing of resolution**
CPU 0
[image]
CPU 1
[image]

I repeated the tests and the traces are the same and repeatable. I am not able to make any sense out of these traces but it seems like one of them triggered the other. Code was compiled using the default with CPU 1.
[image]

Your comments please.

s60sc, 2022-09-28:
Core 1 crashing due to ISR hanging.  The only app ISR is frameISR() - have you modified it? 
Core 0 crash does not trace to the app (as expected) but to freertos and no guru meditation reason is given. 
It is strange for both cores to crash and with very little decoding info so could indicate a hardware problem. do you have peripherals connected or made hardware mods.

YapCK45, 2022-09-28:
Unlikely to be due to hardware problem. Have tested with different ESP32 Cam modules with similar result. The same hardware also worked well using earlier versions eg v6.2.3 and v7.3. 
No modification of ISR. Using the code Out of Box.
I do have pan tilt servos connected to the GPIOs but have not activated them. The web interface is very handy. The Pan Tilt work quite well in v7.3 although have not stress tested them. No other peripheral.

s60sc, 2022-09-28:
try disconnecting the servos from the gpio pins

YapCK45, 2022-09-28:
The servos are not connected to the ESP32 Cam during troubleshooting. Totally floated and disconnected. The ESP32 Cam is only connected to the com port for the serial monitor.

s60sc, 2022-09-28:
As I cant recreate the problem and no one else has reported it, I cant do any further diagnosis

YapCK45, 2022-09-28:
I looked at the code for frameISR() after your comment. Noticed that line 95 in mpeg2sd.cpp is:

timer3 = timerBegin(3, 8000, true); // 0.1ms tick

This assumes that the CPU frequency is 80MHz. It won't be 0.1ms tick if the clock is 240MHz. How would this affect the interrupt frequency?

s60sc, 2022-09-28:
[The timer uses the APB_CLK clock source (typically 80 MHz),](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system_time.html)
The hanging ISR is a freertos ISR